micro-ROS/freertos_apps #5: Olimex E407 Default ping_pong custom_serial uxr_create_session function fails

Kelly-Coffey, 2020-06-05:
ping_pong is unmodified from freertos_apps/apps/ping_pong/ (dashing branch)
To configure I used...
ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial --dev 3
It builds.

I found in rmw_init.c, 
"(!uxr_create_session(&context_impl->session))" 
always returns RMW_RET_ERROR when I configured with transport = custom_serial and dev = 3.

stepping through "uxr_create_session" it steps through by some functions take very long time in my debugger.

attached is an image of the struc session when it is called. 

I have verified transport = udp and app-colon.meta modified to support udp the app builds and works with an Agent, so I think this is something specific with default serial-custom transport.

I got UDP working on another STM32 board and was testing serial mode from completeness. I found this issue on my board and then I reproduced this issue on the supported Olimex E407 hardware.

On another note I forked micro-ros-build and freertos_apps to add STM32F439ZI-Nucleo board support. Is this something you are interested for mainline, or is there a place for other hardware to be posted (ie my git) or not interest at this time?

[image]

Kelly-Coffey, 2020-06-25:
I found another issue where in main.c 
#ifdef MICRO_XRCEDDS_CUSTOM
  availableNetwork = true;
MICRO_XRCEDDS_CUSTOM is not define and it looks like it should be MICRO_XRCEDDS_CUSTOM_SERIAL since when I change it at least the LED flashes the correct rate for link up.

For the 

I went one later deeper and in session.c
    bool received = wait_session_status(session, create_session_buffer, ucdr_buffer_length(&ub), UXR_CONFIG_MAX_SESSION_CONNECTION_ATTEMPTS);
hang up for a while until it hit

    RMW_SET_ERROR_MSG("failed to create node session on Micro ROS Agent.");
    return RMW_RET_ERROR;

I am trying to figure out what is not setup correctly for the custom serial mode.

UDP still works.

FranFin, 2020-07-20:
@Kelly-Coffey Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

Kelly-Coffey, 2021-01-04:
I have this above mentioned configuration with Foxy with both old and new configurations per issue #40 (https://github.com/micro-ROS/freertos_apps/issues/40)  and uxr_create_session does not hang any longer.

closed 2021-01-04 by Kelly-Coffey