micro-ROS/freertos_apps #7: Porting to a Nucleo F446ZE board. App: Ping Pong. Transport: serial uart3

alsaibie, 2020-06-26:
## Setup
Board: Nucleo 446ZE
RTOS: FreeRTOS
App: Ping Pong
Flash Utility: ST-Flash
Debug Utility: OpenOCD
Freertos_apps fork [branch](https://github.com/alsaibie/freertos_apps/tree/dashing/microros_nucleo_f446ze_extensions)
Micro_ros_setup config [branch](https://github.com/alsaibie/micro-ros-build/tree/dashing_stm32/micro_ros_setup/config/freertos/nucleo_f446ze)

## Porting Stage
- [x] builds 
- [x] flashes
- [x] connects and is registered via the linux host microros agent.
- [ ] register ros topic 
- [ ] ping pong  

## Issue
A call to allocator->deallocate in node.c line 412, during ping_pong app node initialization results in a HardFault.
`allocator->deallocate(node_secure_root, allocator->state);`
[image]
The following is a snapshot of the callstack & local variables at the above call
[image]

previous calls to `allocator->deallocate` during node init are executed fine.

Any idea where/what to look for?

pablogs9, 2020-06-26:
Could you check what is the behavior when deallocating an already deallocated memory? I'm not sure right now. Anyways it shouldnt be deallocating an non-allocated (or already deallocated) memory chunk...  Let me check meanwhile

alsaibie, 2020-06-26:
node_secure_root is null. The call to deallocate(node_secure_root =0,0) fails, specifically at the later call to `pxLink->xBlockSize` in [custom_memory_manager.c](https://github.com/alsaibie/freertos_apps/blob/24c342029110fc2f96ff6e78883ea01e60122117/microros_nucleo_f446ze_extensions/Src/custom_memory_manager.c#L291)
`
size_t getBlockSize( void *pv )
{

	uint8_t *puc = ( uint8_t * ) pv;
	BlockLink_t *pxLink;

	puc -= xHeapStructSize;
	pxLink = ( void * ) puc;

	size_t count = pxLink->xBlockSize & ~xBlockAllocatedBit; <--------- 

	return count;
}
`

pablogs9, 2020-06-26:
Well we have to avoid this deallocation. You can fix it by check the null pointer here: https://github.com/alsaibie/freertos_apps/blob/24c342029110fc2f96ff6e78883ea01e60122117/microros_nucleo_f446ze_extensions/Src/allocators.c#L14

But for sure I'm going to check what is happening in the RCL with this `node_secure_root`.

alsaibie, 2020-06-26:
Thanks. Indeed, checking for NULL on node_secure_root in rcl `node.c` or during call to [deallocate](https://github.com/alsaibie/freertos_apps/blob/24c342029110fc2f96ff6e78883ea01e60122117/microros_nucleo_f446ze_extensions/Src/allocators.c#L14) works. The app moves forward and I get the ping and pong topics on the agent, but no messages sent or echoed. I get another HardFault during call to `pxLink->xBlockSize` at reallocate call shown by the following callstack, the xBlockSize is again equal to 0. Bypassing this reallocation call doesn't initialize the `wait_set`
[image]

I notice the puc ptr underflows after `puc -= xHeapStructSize;` as well, doesn't seem right. I'm still trying to wrap my head around the memory manager.
[image]

pablogs9, 2020-06-27:
Ok, so I will check this all in the RCL. Could you PR these memory management changes to our repo? Also, when you have a final version of the port to your board, could you PR the build system and freeRTOS apps changes to our repos? That would be awesome to increase our board support.

Thanks!

pablogs9, 2020-08-06:
Hello @alsaibie I have been checking this, and in the Foxy RCL it seems that this `node_secure_root` has been resolved. 

Can we close this?

Thanks!

closed 2020-08-07 by alsaibie