micro-ROS/freertos_apps #17: Porting to a Discovery1-STM32F407G (FreeRTOS)

RaKalel, 2020-08-05:
Hi everyone !
I am working on porting stm32f407g-disc1 in freeRTOS. For the moment I have successfully configured,built and flashed uros in my board  but I am not able to connect and register with my host microros agent. Do you have an idea where the problem might be ? (I have no error code but I do have some warnings related to allocator.c)

Board: Discovery1 - STM32F407G
RTOS: FreeRTOS
App: int32_publisher
Config, build, flash are working
I am using UART 2 which is connected to my computer by a UART-USB bridge.

Any help would be appreciated.

pablogs9, 2020-08-05:
Hello @RaKalel, can you share the fork where you have implemented the board support? I guess that we should see which transports have you configured.

Some questions:
- Which is your use-case? 
- Where are located you board support files for micro-ROS

RaKalel, 2020-08-05:
Thanks for your reply.
Here is the link for my board support files,
https://github.com/RaKalel/freertos_apps.git

As for the use case, I am a my final year engineering student in Paris and we are trying to check the feasibility of using microros on a specific board to control a set of motors for our cobot .

RaKalel, 2020-08-05:
Hello once again,
My app int 32 pub is working and it is registering with my host agent. In addition the topics are being created without error.
For the moment all good :)

pablogs9, 2020-08-06:
Happy to hear this, if you have any other doubt don't hesitate to open a new issue. I'm closing this since you achieved your goal.

**Please, if you have developed the board support for this new platform, the micro-ROS community will be super happy to integrate it in the repos with a PR.**

closed 2020-08-06 by pablogs9

RaKalel, 2020-08-06:
Hi @Pablogs9,
I am a final year robotics student working on implementing a Micro-ROS + Odrive solution to control a BLDC motor using ROS2. To that end I have been working to port µ-ros on a new board.

I am more than happy to contribute to the micro-ROS community by adding this additional board to the supported list. Here are some more details on my port :
Board: Discovery1 - STM32F407G
RTOS: FreeRTOS
Apps tested: int32_publisher, ping_pong 
I am using UART 2 (serial transport) which is connected to my computer by a UART-USB bridge.
My forks :
 - freertos_apps : https://github.com/RaKalel/freertos_apps.git
 - micro_ros_setup : https://github.com/RaKalel/micro_ros_setup.git
(I am quite new to github so any help is appreciated)

I am able to build, flash, register the client in host and listen to the topic.

I have not been able to perform any memory leak check or proper debugging due to a lack of an IDE (working directly from a terminal).Could you point me towards some good tools existing in Ubuntu that could be used for these types of checks with µ-ros ?

Thank you for your help.

pablogs9, 2020-08-07:
Nice, can you start a pull request from this branches to our repos? Then, there we can start discussing about some changes and integrate this work in the mainline micro-ROS.