micro-ROS/freertos_apps #23: cant create ros nide in upd mode (ESP32)

DorBenHarush, 2020-08-08:
I`ve configured my esp32 with the app int32_publisher in udp mode and got an error while tyring to create a ros node.
the error is this:
UDP mode => ip: 172.17.0.1 - port: 8888 // my docker ip
[ERROR] [0000000028.107394000] [rclc]: [rclc_init] error in rcl_init: failed to create node session on Micro ROS Agent., at /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:306, at /microros_ws/firmware/mcu_ws/uros/rcl/rcl/src/rcl/init.c:229

Failed status on line 37: 1. Aborting.

the line in the code: 
// create init_options
RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

Does anyone know what the problem is?

donRaphaco, 2020-08-12:
Hi @DorBenHarush,
the error means that there the esp can't connect to the agent.
I think the problem is that the IP of your docker is not reachable from other devices within the network (only from your host).
You might need to publish the port of the docker container to the network: https://docs.docker.com/config/containers/container-networking/#published-ports 
Try someting like `-p 8888:8888/udp` and then use the IP of your host PC instead of the docker IP in the `create_firmware` step.

DorBenHarush, 2020-08-12:
It works! Thank you.

closed 2020-08-13 by DorBenHarush