micro-ROS/freertos_apps #24: Mutiple nodes on one board

allstarry, 2020-08-09:
Hi, I want develop something with micro-ros, and I am learning this demo for now. I want to know if micro-ros support mutiple nodes working together, because I didn't see anyting about that. Or it do support it now but need users to some extra work?

ralph-lange, 2020-08-11:
Thank you for your interest in micro-ROS. In principle, micro-ROS supports multiple nodes, but all communication between them will go through the agent. An efficient inter-node communication on the microcontroller is an open issue (cf. https://micro-ros.github.io/docs/overview/features/ -> _Inter- and intra-process communication using the same API_).

allstarry, 2020-08-14:
According to your saying, micro-ROS can support multiple nodes working together but doesn't have an efficient way to communicate with each other, am I right?
If so, I think the tutorial of micro-ROS is way behind, since I haven't found any instruction about how to make multiple nodes work together.
Now I am working on it, but I found this function rclc_support_init() is a bit odd. I found it can only execute once in the whole application, so I tried to move it to the initiation task and pass attributes to the node task, but the whole application just don't work.
Does it need some special condition?

JanStaschulat, 2020-08-14:
Yes, you can have multiple nodes on the micro-controller. What is meant with in-efficient communication is, that assuming you have one publisher and one subscriber on the micro-controller. Then the communication goes via the agent. Our idea is to have local communication on the micro-controller board for thoese cases, in which publisher/subscriber are only on the board.

The rclc_support_init() function calls rcl_init - to initialize the ROS context, and secondly, initializes the ros clock rlc_clock_init() It was ment only for convenience only. So you can use the same 'support' object for multiple nodes. 

Another reason, why it might not work, is, that you have to specify how many publisher/subscriber are supported on the micro-controller application. The configuration file is placed in rmw_microxrcedds_c/rmw_microxrcedds.config and has the following configurable parameters: (https://github.com/micro-ROS/rmw-microxrcedds)
...
RMW_UXRCE_MAX_PUBLISHERS: This value sets the maximum number of publishers.
RMW_UXRCE_MAX_SUBSCRIPTIONS: This value sets the maximum number of subscriptions.

Check that you are not using more pub/subs than configured.

Could you send the source-code of your setup?

allstarry, 2020-08-15:
Actually I'm trying to using multiple nodes in multiple threads. I do noticed that when it comes to micro-controller and RTOS, using multiple threads may be a better way to separate my application, but nodes have their life circle mechanism to make their management easier.
For this freertos_apps, I always modify the app-colcon.meta in the apps folder instead of configure file of  microxrcedds, and I'm using less publishers and subscribers than the demo application, so I don't think it's the reason either.
I'm just modifying some code of demo, so my setup is very simple:
1.Copy the app.c file and delete a publisher and all the subscriber then change the thread name, so I have two new threads and one publisher for each.
2.Move allocator and support initiation to the initTaskFunction, and pass the pointers of them to the new threads .
3. Delete the allocator and support initiation in each thread and accept the pointer passed from initTaskFunction.
That's what all I have done.According to the printf output, I can confirm that both tasks has been started, but only one publisher can work, and the other thread stopped when this error occured: [rclc_node_init_default] error in rcl_node_init: error not set Failed status on line 83: 1. Aborting

JanStaschulat, 2020-08-17:
Hi @allstarry, 
unfortunatey the current version of micro-ROS xrce-dds only supports single threaded applications on the micro-controller. If you need multiple threads then do all ROS communication (publish/subscribe) in one thread and use a semaphore to exchange data between threads. We did this in the kobuki demo.

  @pablogs9 , what are the current plans for the xrce-dds middleware to support multithreading?

allstarry, 2020-08-18:
Thanks for your explanation and solution, and I'm looking forward to the implementation of this feature.

BTW, I noticed that subscribers can be a very big occupation of RAM, so the micro-ROS team may be better to put up a list about how many functions of micro-ROS you can have for each hardware platform due to we always have limited resources on board and a list about the popular feathers this project can support for now as well.

Best regards.