micro-ROS/freertos_apps #28: Adding a new board to use with freeRTOS

edesta-be, 2020-09-11:
Hi are there a set of steps we can follow to build a new board to run with freertos and uros?
For your reference I am using the nucleo_f746zg. I tried to imitate the f4 example you all already have in the example but I got stuck at the extensions and how to continue after that point. 

Thank you
Look forward to your response.

pablogs9, 2020-09-14:
Hello @edesta-be, by now micro-ROS FreeRTOS for STM32 ports relies on CubeMX, could you tell us how is your environment configured? Do you have a ready to use CubeMX project?

edesta-be, 2020-09-14:
Ohh cool! I was not aware. Essentially I have initialised all my peripherals and have had the cube generate the code. Thats where I am, I havent implemented anything complex. I just realised this board was not supported when i tried to configure an app.

If this is the case @pablogs9 is there anywhere i can refer to?

pablogs9, 2020-09-14:
No, we don't have a tutorial for implementing a new board in FreeRTOS. But, you can see that some of this `_extension` folders are just CubeMX project [check this](https://github.com/micro-ROS/freertos_apps/tree/foxy/microros_olimex_e407_extensions).

A good starting point for porting micro-ROS+FreeRTOS to a new board is to check the differences between your generated project and one of these folders. You will find that [Makefile](https://github.com/micro-ROS/freertos_apps/blob/b666c692355169ebc88437aa2fe47c62ad7bc572/microros_olimex_e407_extensions/Makefile#L316) has some colcon related parts, you also have the init task [launching](https://github.com/micro-ROS/freertos_apps/blob/b666c692355169ebc88437aa2fe47c62ad7bc572/microros_olimex_e407_extensions/Src/main.c#L417) the micro-ROS app.

Another important part is the [FreeRTOS+POSIX extension](https://github.com/micro-ROS/freertos_apps/tree/foxy/microros_olimex_e407_extensions/FreeRTOS-Plus-POSIX), you have to add it in order to have micro-ROS successfully built. 

Please analyze the differences between this folder and yours so we can help you with this issue. All new board ports are really welcome.

edesta-be, 2020-09-14:
Hi @pablogs9 Thank you for your response!
So I was imitating your implementation of the nucleo_f446ze.
I notice that this board does not have a freeRTOS+POSIX extension. 
Would you recommend me imitating the Olimex board and getting back to you or the nucleo_f446ze board?

pablogs9, 2020-09-14:
Hello! Nucleo F446ZE has the POSIX interface in another folder. [Check here](https://github.com/micro-ROS/freertos_apps/tree/foxy/microros_nucleo_f446ze_extensions/lib/FreeRTOS-Plus-POSIX). Anyways, this is a [community contribution](https://github.com/micro-ROS/freertos_apps/pull/11), I haven't tested it.

So I recommend the Olimex one because is the one we develop in the micro-ROS team.

edesta-be, 2020-09-14:
Perfect! Thank you @pablogs9

pablogs9, 2020-09-14:
BTW, we have been thinking in publishing a tutorial in [micro-ROS webpage](https://micro-ros.github.io//docs/tutorials/core/overview/) on how to do these kinds of FreeRTOS ports. Maybe if is possible for you to write down the steps, later we can join efforts and write it for the rest of the micro-ROS users. On Wednesday we have a ROS 2 Embedded Working Groups and probably I will propose this tutorial topic. Let us know if you want to assist.

Of course, let us know when you stuck some point (providing a branch where we can test the port) and we can solve it together.

pablogs9, 2020-09-15:
[Link to the agenda and the google meet room](https://docs.google.com/document/d/1hGxwpuHDPCyEYHBCd5IIgvoFGJgkC5WKI12ghEfqwXg)

edesta-be, 2020-09-15:
@pablogs9 it seems there are some very annoying nitty gritty updates that we will have to do with any board that isnt using the F4 hal drivers. 
There is also some odd errors i am catching with the update. Let me know how you want me to send them to you.

pablogs9, 2020-09-16:
Hello @edesta-be, regarding the F4 I guess that all supported boards use this STM32 family, could you detail here these updates in order to start discussing how to approach the solution. Please provide an explanation as detailed as you can.

Regarding the F7, the same thing. Let us know these runtime errors, provide a branch and detailed explanation.

Thank you very much for the feedback!

edesta-be, 2020-09-16:
@pablogs9 I am unsure if it is normal and expected, but during the build of the system there are a ton of warnings that could be worrisome in the long run. I didnt put any effort into reducing that number however they do exist with all the builds that have the F4 drivers.

Initially, to get started, in the microROS setup file that we pulled down, I created a new config, named it nucle_f743zg. Most of the files that I ported over were identical to the files found in the olimex folder. I only tweaked small things as i saw naming etc were hardcoded in. 

In terms of the work that I did with the extensions, i walked through setting up an IOC with similar features as the OLIMEX board. The only real things i changed where the DMA, RCC, SYS and ETH, LWIP. The clock was set to the maximum MHz. 

With these updates that imitate yours, I generated a makefile using CubeMX. Then, on my own end i added all the unique files etc that I had to move over (the freeRTOS + POSIX etc). The makefile also had a speicific build that was targeted towards the F4 boards in the microROS setup section, I changed that to f7 accordingly. 

The rest was just following the steps that have been outlines in making any project and building the firmware. 

So, in order to test this system, I configured the board with the int32 publisher and i caught the weirdest error. 
The microros agent seems to catch on something and run really fast without producing anything on the other end. (i was echoing the topic on another terminal making sure everything was working etc.) There would be a catch on that end, and we would stop recieving data. On the agent end you would see the data just filtering at a much faster speed. 
In the same stream, when I press the physical reset button on the board it takes around 10-20 seconds for the board to actually reset. 

Since I had nothing I could compare it against, I decided to implement the F7 board with zephyr and test if it was the board etc (i was just trying to reduce the location of points in failure -- most of my thought being perhaps it was a bad board). When running the same app on zephyr, i noticed that there was no lag and the agent didnt not start running really fast without producing on the other end. 
Again, to reduce potentially anything that might be board specific, i tested the FreeRTOS stuff on a second F7 board, and I noticed the same thing. Rerun the Zephyr app on this board, again no problems. 

Now i can say that it likely exists in one of the points of setup in the project and would love to see if we can go through it together and determine what problems we can nip. 

Thank you!

pablogs9, 2020-09-17:
Please, can you share a log of this micro-ROS agent running fast/with errors? Ensure that you have set the `-v6` flag

I assume that you are using serial transport right?

edesta-be, 2020-09-17:
Yes (v6 and serial), where do I find the error log?

pablogs9, 2020-09-17:
Just put here the output of the agent with the entities creation and so on.

edesta-be, 2020-09-17:
I'm not sure I follow @pablogs9 You want me to show the output of the terminal screen?

pablogs9, 2020-09-17:
yes, I want to see what this "run really fast" means in terms of agent behavior

edesta-be, 2020-09-17:
[freeRTOS_terminal_output.txt](https://github.com/micro-ROS/freertos_apps/files/5240576/freeRTOS_terminal_output.txt)

This the output of a full run with my board (it initially works normally with a small lag when i reset the board -- and then around the end microROS is not even publishing the integers)

pablogs9, 2020-09-18:
Hello @edesta-be, I have seen this behavior before. It seems that works and the client is able to create entities and publish data. But at some point, they get stuck sending ACK and heartbeat (internal session management of the middleware).

I this can be a transport of timing issue, some questions:
- Which transport are you using? Can you provide a link to the file?
- Is this log the `int32_publisher` application? With no modifications? In some cases, the timings (spin time or sleep time in the loop) can cause problems.
- Can you provide information about the serial configuration of your board? Did you set buffers length or something?

jamoralp, 2020-09-18:
Good morning @edesta-be! I've been taking a look at the logs, and it looks very familiar to me; we had this issue in the past due to a bug introduced in the agent, which was fixed some weeks ago. Could you please check which version of the Agent are you  using?

edesta-be, 2020-09-18:
@pablogs9 I'm not sure what you mean by transport, and yes i am running the int32_publisher with no modificiations
i did not change the serial configuration of the board beyond the standard initialisation

@jamoralp that is good to hear that it has been somewhere! I built the ws following the commands on the read me, so i imagine that is the most recent version of the system.

pablogs9, 2020-09-21:
Regarding the transport layer: the very last layer in micro-ROS middleware is implemented in some cases with platform-specific functions. [Check here](https://github.com/micro-ROS/freertos_apps/blob/foxy/microros_olimex_e407_extensions/Src/olimex_e407_serial_transport.c). Four functions are needed: init, close, send and receive, so in this case, you need to guarantee that the application on top of the RTOS is able to write and read from the transport.

Which priority and stack has the [main application](https://github.com/micro-ROS/freertos_apps/blob/8c8baba17e50f428efeb86111a86560221df044e/microros_olimex_e407_extensions/Src/main.c#L417) in your app?

edesta-be, 2020-09-21:
@pablogs9 The serial functions definitely communicate with the application on top. I might be wrong in assuming this but would the application have ran in the first place if it didnt? 
And as for the priority and stack the initFunction attributes are exactly the same as yours and the attributes thread are also setup with the same variables on your end.

@jamoralp We pulled the most recent version of the agent. What was the bug initially and how was it solved?

edesta-be, 2020-09-21:
Hello!

So i made some more progress! It looks like i had not set my DMA RX pin to be Circular, which is an annoying error on my end. This solved the problem of the lag hold up and "going very fast and being unable to transmit". The only problem i am continuing to experience is the time it take to reset. 

When i use the button on the board to reset, it pauses the stream on the agent (so i know it is beginning to reset) and then after approximately 6 seconds it starts up the output to the terminal (via the agent). I'm not sure if i am missing something fundamental on that end. But after that i think we can successfully coin this as a good transport over.

pablogs9, 2020-09-22:
Hello @edesta-be, really nice to hear about this progress!!! This ACK-Hearbeats fast interchange usually is due to transport issues. Are you thinking in incorporating support to your board with a PR to our repositories? 

Regarding the 6 seconds startup time, have you profiled which function is taking this time? Usually depends on the transport, but in a serial mode it shouldnt take so long...

edesta-be, 2020-09-22:
Hello @pablogs9 I have not profiled which function, I can do that next and see what the hold up is. I was thinking that it should not take that long so it is good to have that confirmed. Do you have a general time it takes to flash the olimex board? And the general number of bytes that are written when you flash the board? 

The flash on my end: "wrote 262144 bytes from file micro-ROS.bin in 6.104579s (41.936 KiB/s)"

And yes! I do hope to put in the board support with a PR!

pablogs9, 2020-09-22:
But, we are talking about the flashing time or the boot time?

We have not measured the flashing time, but 6 seconds sounds good for me because the JTAG has to erase and rewrite the whole memory. 262 kB of flash memory also sounds ok.

I was talking about the boot time of the board once it is already flashed, for this I guess that micro-ROS should take less than 6 seconds to communicate with the agent, but it would be nice to know in which function are these 6 seconds spent.

edesta-be, 2020-09-22:
Yes! So, it appears that once i removed the LWiP init, the board boot up decreased. It is practically instantaneous. Which make me think it had something to do with that instead. I am likely going to call that init asynchronously so it does not cause the holdup on reset. 

Thank you so much! 
How do i go about putting in a PR for this board. And I will likely be opening a new issue for the custom app support that you did for Zephyr. It looked like you all only affected two files which I am more than happy to do in another PR if need be. Let me know.

Sincerely, 
Eden

pablogs9, 2020-09-23:
For this board support with FreeRTOS, we need to have two PR:

- One of them in this repo adding the `extensions` folder. You can take this one as reference: https://github.com/micro-ROS/freertos_apps/pull/11
- Other in [micro_ros_setup](https://github.com/micro-ROS/micro_ros_setup), please take this one as reference: https://github.com/micro-ROS/micro_ros_setup/pull/139

So, can we close this issue? Feel free to open a new one for this Zephyr issue.
Thanks!

edesta-be, 2020-09-24:
Thank you! Will do!

closed 2020-09-24 by edesta-be