micro-ROS/freertos_apps #30: Porting micro-ROS on Zynq FPGA

yongyi1837, 2020-09-24:
Hello,
I'm trying to do the tutorial 'First micro-ROS Application on FreeRTOS' using vivado with C language. But I have no idea how to translate the terminal-instructions to a program, or how to realize the functions of porting micro-ROS on FreeRTOS in code. Could anyone give me some suggestions? Keywords and documentations?
Thank you.
Best regards,
YongYi

pablogs9, 2020-09-25:
Hello, Vivado is not currently supported in the micro-ROS build system and AFAIK the implementation would be completely different from the current gcc-based solution.

Anyways this port seems to be really interesting, could you provide more details about this software build system. Maybe it is possible to have a CMake toolchain and use [this new experimental tool](https://github.com/micro-ROS/micro_ros_setup/pull/190) to generate static libraries outside Vivado and the link against them?

Please provide more details about the toolchain and build system of your Vivado+FreeRTOS project.

yongyi1837, 2020-09-25:
Hello,
Under vivado it builds a hardware diagram like this:
[image]

And then I've chosen 'freertos10_xilinx' for OS platform under SDK.

Are they the toolchain and build system what you mentioned?

Best regards,
YongYi

pablogs9, 2020-09-25:
Do you have access to the source code of the application been built for the ARM core?

yongyi1837, 2020-09-25:
> Do you have access to the source code of the application been built for the ARM core?

I think I don't have the access for ARM core.

yongyi1837, 2020-09-25:
[image]

The FreeRTOS is built automatic after I have set 'freertos10_xilinx' for OS platform.
The thing is how to realize porting micro-ROS on this in program. I have no idea where to start and what I need.

pablogs9, 2020-09-29:
Hello, the tutorial you are following does not explain how to integrate micro-ROS in a new platform, it explains how to test micro-ROS in the supported boards.

If  you want to port micro-ROS to a new platform you have to integrate the micro-ROS build system in your target build system, in the case of FreeRTOS please have a look to this files to see how micro-ROS has been integrated with STM32CubeMx for our reference platform Olimex STM32 E407: https://github.com/micro-ROS/freertos_apps/tree/foxy/microros_olimex_e407_extensions

yongyi1837, 2020-10-02:
I have created a project with SDK, that is bilt with FreeRTOS and lwIP. Is that mean, I just need to add "allocators.c", "custom_memory_manager.c", "ethernetif.c" and "main.c"? Or should I add other files in Src too?

pablogs9, 2020-10-02:
These are extra files included to the FreeRTOS installation. The point where micro-ROS is built is here: https://github.com/micro-ROS/freertos_apps/blob/34656bb8e1b5f80a2ef4f05e26da371f7538da55/microros_olimex_e407_extensions/Makefile#L325

You should be able to integrate this into your Makefile

closed 2020-11-16 by pablogs9