micro-ROS/freertos_apps #31: Custom App Support W/ FreeRTOS

edesta-be, 2020-09-24:
Could we also have a Custom App Feature similar to what was done for zephyr (https://github.com/micro-ROS/micro_ros_setup/pull/183) 

thank you!

pablogs9, 2020-09-25:
Hello @edesta-be, great idea. I don't know if you have dug inside other RTOS build systems for micro-ROS but in the case of Nuttx and Zephyr there is a unified build system, in the case of FreeRTOS each platform has their own scripts. [Here you can see this differentiation](https://github.com/micro-ROS/micro_ros_setup/tree/foxy/config/freertos).

Anyways I think that this is completely possible to implement, what do you think about creating a PR with your port so we can work together there?

I will put some commits in this PR with these changes regarding `UROS_CUSTOM_APP_FOLDER`.

edesta-be, 2020-09-25:
Okay sounds good! I will be putting in a PR soon, i'm just cleaning some things up on my end

edesta-be, 2020-09-28:
Hi @pablogs9 added a pull request for freeRTOS apps and the setup.
Just on top of that i went ahead and added support for this board with zephyr because it was pretty simple.
I affected the setup and the zephyr apps for that as well. So i have three PRs open. Let me know if I am missing anything and I would be happy to fix it!

Thank you for all your help!

pablogs9, 2020-09-29:
Hey @edesta-be, closing here and moving the conversation to https://github.com/micro-ROS/micro_ros_setup/pull/197

closed 2020-09-29 by pablogs9