micro-ROS/freertos_apps #33: Cannot trace pingpong on OlimexE407-_FreeRTOS

Aerotic, 2020-10-21:
I'm so sorry. I don't know a proper repo where to submit this issue.

HOST OS: Ubuntu 20.04.1 LTS

used command for tracing: 
```shell
ros2 trace
```
error log:
```shell
aero@aero:~/ros/freertos$ ros2 trace
UST tracing enabled (16 events)
kernel tracing enabled (4 events)
context (3 names)
writing tracing session to: /home/aero/.ros/tracing/session-20201021211759
press enter to start...
channel enabling failed: Tracing the kernel requires a root lttng-sessiond daemon, as well as "tracing" group membership or root user ID for the lttng client.
```
 I'v done everything mentioned in the tutorial https://micro-ros.github.io/docs/tutorials/advanced/tracing
but i'm stuck at this step `ros2 launch tracetools_launch example.launch.py`, which report similar error about  `Tracing the kernel requires a root lttng-sessiond daemon, as well as "tracing" group membership or root user ID for the lttng client`

pablogs9, 2020-10-21:
Hello @Aerotic,

I'm not sure about the status of ros2tracing in micro-ROS. I can't provide information about the scope of this tutorial, maybe @ralph-lange or @iluetkeb can clarify something here.

Also, maybe someone [here](https://gitlab.com/ros-tracing/ros2_tracing) can provide some detailed information.

ralph-lange, 2020-10-21:
Hi @Aerotic, thank you for reaching out. I just tried the tutorial on a fresh Ubuntu 20.04 installation with ROS 2 Foxy and it worked for me immediately. Yet, one detail is not mentioned in the tutorial: You have to add your user to the group _tracing_ (by `sudo usermod -aG tracing $USER`). Please note that you have to log off and log in again to update the group memberships before calling `ros2 launch tracetools_launch example.launch.py` again.
Does this solve the issue?

iluetkeb, 2020-10-21:
Thanks to @ralph-lange for helping out!

btw, the main tracetools repository is over at https://gitlab.com/ros-tracing

Aerotic, 2020-10-21:
> Hi @Aerotic, thank you for reaching out. I just tried the tutorial on a fresh Ubuntu 20.04 installation with ROS 2 Foxy and it worked for me immediately. Yet, one detail is not mentioned in the [...]

Thanks to your solution. I'll try it as soon as possible!

Aerotic, 2020-10-21:
> Thanks to @ralph-lange for helping out!
> 
> btw, the main tracetools repository is over at https://gitlab.com/ros-tracing
I'm so sorry.
Should I resubmit this issue to ros-tracing or micro-ROS.github.io ?

pablogs9, 2020-10-21:
I'm moving this issue to micro-ROS.github.io.