micro-ROS/freertos_apps #34: fatal error: lttng/tracepoint.h: No such file or directory

Aerotic, 2020-10-26:
HOST OS: Ubuntu 20.04.1 LTS
Board: Olimex E407
RTOS: FreeRTOS

I‘d like to use the `tracetools`
So I modify something in `colcon.meta` (PATH `firmware/mcu_ws/colcon.meta`):
```
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=OFF", # the previous is ON
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=ON" # the previous is OFF
            ]
        },
```

But I got errors when I run `ros2 run micro_ros_setup build_firmware.sh` :
```
/home/aero/ros/freertos/firmware/mcu_ws/build/tracetools/include/tracetools/tp_call.h:29:10: fatal error: lttng/tracepoint.h: No such file or directory
 #include <lttng/tracepoint.h>
          ^~~~~~~~~~~~~~~~~~~~
```

```
/home/aero/ros/freertos/firmware/mcu_ws/uros/tracetools/tracetools/src/utils.cpp:20:10: fatal error: dlfcn.h: No such file or directory
 #include <dlfcn.h>
          ^~~~~~~~~
compilation terminated.
```

I can find both the two header files in my computer:
`/usr/include/dlfcn.h` and `/usr/include/x86_64-linux-gnu/lttng/tracepoint.h`

Can someone help me ?

pablogs9, 2020-10-26:
At this moment micro-ROS port for FreeRTOS does not have official support for tracetools. We have not tested if it builds nor if it works. 

Suggestions on how to make it works or contribution PR are welcome.

closed 2020-10-26 by pablogs9

Aerotic, 2020-10-26:
> At this moment micro-ROS port for FreeRTOS does not have official support for tracetools. We have not tested if it builds nor if it works.
> 
> Suggestions on how to make it works or contribution PR are welcome.

Thanks for your reply:blush: