micro-ROS/freertos_apps #35: Publishing from different FreeRTOS task does not work

dsryzhov, 2020-10-27:
Hi,

When trying to publish messages from different FreeRTOS task  errors occur
and esp32 going to reboot

I've found the same error description with esp32 web server
https://www.esp32.com/viewtopic.php?t=3198

Errors of two types occur

1. 
CORRUPT HEAP: multi_heap.c:395 detected at 0x3ffbf968
abort() was called at PC 0x40086955 on core 1

2. 
INFO: rcl_wait timeout 0 ms
assertion "pbuf_free: p->ref > 0" failed: file "/opt/esp/idf/components/lwip/lwip/src/core/pbuf.c", line 757, function: pbuf_free
abort() was called at PC 0x40142bab on core 1

ELF file SHA256: e47e78ae1b6cd29d

Backtrace: 0x40087901:0x3ffc8e10 0x40087c85:0x3ffc8e30 0x40142bab:0x3ffc8e50 0x400f189b:0x3ffc8e80 0x40102e19:0x3ffc8ea0 0x400edf63:0x3ffc8ec0 0x400eeb6e:0x3ffc8f00 0x400eebee:0x3ffc8f60 0x400e4021:0x3ffc8f80 0x400e3e3a:0x3ffc8fb0 0x400def76:0x3ffc8fe0 0x400df678:0x3ffc9080 0x400dcf10:0x3ffc90a0 0x400d9571:0x3ffc90f0 0x400d612d:0x3ffc9110 0x40152221:0x3ffc9130 0x400d8985:0x3ffc9150 0x40088215:0x3ffc9180

Rebooting...
ets Jun  8 2016 00:22:57

If some information is available on this error please share it. 

Regards,
Dmitry

dsryzhov, 2020-10-27:
Use case for publishing from other task:

I am trying to publish sensor data every time it's measured from different task.
If data is published by timer (from Micro-ROS task( there is a a time difference between measurements and publishing
that I want to minimize.

pablogs9, 2020-10-27:
Hello @dsryzhov micro-ROS is not prepared for multithreading yet since its middleware (Micro XRCE-DDS Client) is a single thread library.

I suggest you have a single micro-ROS task that communicates with your other tasks using FreeRTOS utilities such as queues.

Regarding your last comment, please could you share some code in order to suggest the best approach for your use case?

dsryzhov, 2020-10-27:
Hello @pablogs9!
Thanks for your quick response! Now I know that this behaviour is not a bug but feature. 

As I understand receiving events from a FreeRTOS queue could be handled in a loop with rclc_executor_spin_some.
In a loop
  - rclc_executor_spin_some
  - handle events from FreeRTOS queue (from other tasks) and publish sensor messages

In this case time difference betwen sensor valed measured time and published time could be smaller than publishing in a regular time interval. 

I will try to implement this behaviour and share my result code.

Regards,
Dmitry

pablogs9, 2020-10-27:
That's it @dsryzhov. Looking forward to seeing your micro-ROS applications!

I'm closing this, do not hesitate to reopen if you have any other doubt.

closed 2020-10-27 by pablogs9