micro-ROS/freertos_apps #37: Package 'rcutils' exports library 'dl' which couldn't be found

ncnynl, 2020-11-14:
I build the ping_pong app allways have the CMake Warning : 

Package 'rcutils' exports library 'dl' which couldn't be found 

I can compile normally, get the bin file, or burn it to the board,
and the LED light flashes normally. 

The  detail infomation as below: 

Starting >>> rosidl_typesupport_microxrcedds_c
--- stderr: rosidl_typesupport_microxrcedds_c
CMake Warning at /home/ubuntu/microros_ws/firmware/mcu_ws/install/rcutils/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /home/ubuntu/microros_ws/firmware/mcu_ws/install/rcutils/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /home/ubuntu/microros_ws/firmware/mcu_ws/install/rosidl_runtime_c/share/rosidl_runtime_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/ubuntu/microros_ws/firmware/mcu_ws/install/rosidl_runtime_c/share/rosidl_runtime_c/cmake/rosidl_runtime_cConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -b 115200 -v 6

[1605340011.161238] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1605340011.161433] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1605340015.254941] info     | Root.cpp           | create_client            | create                 | client_key: 0x5851F42D, session_id: 0x81
[1605340015.255052] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1481765933, address: 1

But cannot display ros2 topic information   /microROS/pong

$ ros2 topic list

/parameter_events
/rosout

RobertWilbrandt, 2020-11-14:
These two problems seem unrelated. I already wanted to take a look into the rcutils dt warning, as it happens in every mcu_ws build for me too (basically since forever). But this does not cause any problems in program behavior.

ncnynl, 2020-11-15:
Maybe my understanding of micro ros is not enough. At present, I have changed the source code of freertos based on olimex-stm32-e407 and interacted through UART1. Currently, relevant information can also be output to the UART1 serial port, but the topic cannot be output normally.

pablogs9, 2020-11-16:
Hello @ncnynl:

Regarding the `dl` library, it is a dynamic library loader user un Linux platforms and required optionally for some of the ROS 2 packages which are also used in micro-ROS. Of course in embedded scenarios, we don't have dynamic library loading, so this library is not needed.

Regarding your ping pong application, your agent is not creating entities. You have the verbosity enabled (`-v6`) but it doesn't not show any debug trace of DDS entities being created. Could you share your procedure so we can reproduce the issue here?

ncnynl, 2020-11-20:
@pablogs9   have fixed it   !   The problem is the same as the other one  #38

closed 2020-11-20 by ncnynl

pablogs9, 2020-11-20:
The domain ID?

I'm closing sice solved

roncapat, 2021-10-10:
I'm sorry but I have the same warning. Any fix? See https://github.com/micro-ROS/micro_ros_setup/issues/397