micro-ROS/freertos_apps #38: Adding a new board to use with freeRTOS ， but can not get the topic on other console

ncnynl, 2020-11-17:
I have a stm32f407 board with the same chip as olimex_e407, I want to transplant the source code of olimex_e407 to my new version. I follow the tutorial https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/ to complete the related installation and burning. My problem is that I cannot see the topic /microROS/ping in the new terminal.

I start the agent as follows:

ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6

[1605573834.109076] info | TermiosAgentLinux.cpp | init | running... | fd: 3
[1605573834.109228] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 6

After I restart the device, I can see the serial port transceiver and dds reading and writing:

[1605573865.009277] debug | SerialAgentLinux.cpp | recv_message | [==>> SER <<==] | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 0B 00 0B 00 80
[1605573865.009292] debug | SerialAgentLinux.cpp | recv_message | [==>> SER <<==] | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 0B 00 0B 00 80
[1605573865.009370] debug | DataWriter.cpp | write | [** <<DDS>> **] | client_key: 0x00000000, len: 34, data:
0000: 04 00 00 00 00 6E 0A 1E 16 00 00 00 31 31 39 31 33 39 31 35 32 39 5F 31 30 38 35 33 37 37 37 34
0020: 33 00
[1605573865.009428] debug | DataReader.cpp | read_fn | [==>> DDS <<==] | client_key: 0x00000000, len: 34, data:
0000: 04 00 00 00 00 6E 0A 1E 16 00 00 00 31 31 39 31 33 39 31 35 32 39 5F 31 30 38 35 33 37 37 37 34
0020: 33 00
[1605573865.009447] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 0C 00 00 00 80
[1605573865.009516] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 0C 00 00 00 80
[1605573865.009531] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 0C 00 00 00 80
[1605573865.009555] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x5851F42D, len: 46, data:
0000: 81 01 01 00 09 01 26 00 00 14 00 06 04 00 00 00 00 6E 0A 1E 16 00 00 00 31 31 39 31 33 39 31 35
0020: 32 39 5F 31 30 38 35 33 37 37 37 34 33 00
[1605573865.010425] debug | SerialAgentLinux.cpp | recv_message | [==>> SER <<==] | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 0B 00 0B 00 80
[1605573865.010551] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 0C 00 00 00 80

But I only see

 $ ros2 topic list
/parameter_events
/rosout

I can't see it normally, the topic is like this
/microROS/ping
/microROS/pong

I have no clue what the problem is. . Do you have any suggestions?

The serial port source code I use is the same

pablogs9, 2020-11-17:
Please paste the complete trace of the micro-ROS Agent.

ncnynl, 2020-11-17:
@pablogs9

ubuntu@AiROS2-NUC:~$ cat trace_agent.log
[1605598119.134960] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1605598119.135195] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1605598121.774195] info     | Root.cpp           | create_client            | create                 | client_key: 0x5851F42D, session_id: 0x81
[1605598121.774369] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1481765933, address: 1
[1605598121.774558] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 19, data:
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1605598121.807071] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 100, data:
0000: 81 80 00 00 01 05 5C 00 00 0A 00 01 01 02 00 00 4E 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 70 69 6E 67 70 6F 6E 67 5F 6E 6F 64 65 3C
0040: 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73
0060: 3E 00 00 00
[1605598121.807151] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1605598121.814775] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x5851F42D, participant_id: 0x000(1)
[1605598121.814978] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data:
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1605598121.815011] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1605598121.815024] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1605598121.863361] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1605598121.863437] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1605598121.863454] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1605598121.863469] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1605598121.863519] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 128, data:
0000: 81 80 01 00 01 05 77 00 00 0B 00 02 02 02 00 00 69 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 6D 69 63 72 6F 52 4F 53 2F 70 69 6E 67 3C 2F 6E 61 6D 65 3E 3C 64 61
0040: 74 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 48 65 61 64
0060: 65 72 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00
[1605598121.863541] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 01 00 80
[1605598121.863591] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1605598121.863612] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1605598121.863629] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1605598121.863633] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1605598121.863696] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1605598121.863709] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1605598121.863824] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x5851F42D, topic_id: 0x000(2), participant_id: 0x000(1)
[1605598121.863955] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data:
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1605598121.863985] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[... cut ...]
[1605598128.213410] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 0D 00 00 00 80

I did some other attempts:

1. Tested and reinstalled foxy from source code, but it didn't work.
2. Testing the example at https://micro-ros.github.io/docs/tutorials/core/first_application_linux/ is no problem. 
3. The master version and v1.5.0 version of Micro-XRCE-DDS-Agent have been replaced. but it didn't work.

pablogs9, 2020-11-17:
It seems that your agent is publishing to DDS so I want to ensure that your `ros2 topic list` is able to listen to this DDS dataspace.

---

Please, use our default versions of the repos. Try a micro-ROS Agent with:

```bash
docker pull microros/micro-ros-agent:foxy
docker run -it --rm --net=host --privileged -v /dev:/dev microros/micro-ros-agent:foxy serial --dev [YOUR DEV] -v6
```

Then check it with:

```bash
docker pull ros:foxy
docker run -it --rm --net=host ros:foxy
# Inside the docker
ros2 topic list
```

ncnynl, 2020-11-17:
@pablogs9   

Thanks, i try

ncnynl, 2020-11-17:
@pablogs9 

Using docker, it can work normally.

```

ubuntu@AiROS2-NUC:~$ sudo docker run -it --rm --net=host ros:foxy
root@AiROS2-NUC:/# ros2 topic list
/microROS/ping
/microROS/pong
/parameter_events
/rosout

```

```
root@AiROS2-NUC:/# ros2 topic echo /microROS/ping
stamp:
  sec: 260
  nanosec: 495000000
frame_id: '2011243547_1085377743'
---
stamp:
  sec: 262
  nanosec: 494000000
frame_id: '241418521_1085377743'
---
stamp:
  sec: 264
  nanosec: 501000000
frame_id: '699304830_1085377743'
---
stamp:
  sec: 266
  nanosec: 497000000
frame_id: '261159140_1085377743'

```

That means there is a problem with my foxy environment configuration, and the code migration of the development board has been working normally.

pablogs9, 2020-11-17:
I think so. 

Have you changed the `DOMAIN_ID` ?
Have you updated recently the foxy installation in your host?

ncnynl, 2020-11-17:
@pablogs9  

Thank you pablogs9  for helping me solve the big problem
Because I used burger before, I set the following parameters
#export ROS_DOMAIN_ID=30
#export TURTLEBOT3_MODEL=burger
I comment these parameters and test again. It will work normally.

pablogs9, 2020-11-17:
Nice, I'm closing this since solved. 

Remember that you can also use different domains ID with micro-ROS. [Check this](https://github.com/micro-ROS/micro-ROS-demos/blob/24df6830734bb43078e60b4ae0c10a9feb09d3ef/rclc/configuration_example/configured_publisher/main.c#L52).

closed 2020-11-17 by pablogs9

ncnynl, 2020-11-17:
Okay, I will read the code more, thanks again