micro-ROS/freertos_apps #39: cannot subscribe to best effort publisher

cocodmdr, 2020-11-23:
Hello,

I am using microROS since two months now and I noticed some strange behaviour.

I am running the ping_pong example with freeRTOS on olimex board.
Everything runs fines when publishing fake ping I can see a pong coming.

However I cannot subscribe to /microROS/pong messages in ROS.

- I can see the pong with the built-in ROS2 tool : **ros2 topic echo /microROS/pong**
- But I cannot subscribe to this topic with a simple subscriber like this one : https://index.ros.org/doc/ros2/Tutorials/Writing-A-Simple-Cpp-Publisher-And-Subscriber/

I noticed that it is related to this function  **rclc_publisher_init_best_effort()**

Changing this function with **rclc_publisher_init_default()** everything works well and I can subscribe to the topic.

pablogs9, 2020-11-23:
Could you please do a `ros2 topic info [topic name]  -v` having both the Cpp app and the micro-ROS app up and running and paste here the output? 

It seems that there it some kind of QoS misunderstanding.

pablogs9, 2020-11-23:
I'm moving this to the micro_ros_setup repo, since it is related to the whole micro-ROS