micro-ROS/freertos_apps #40: UDP transport fails after some time

cocodmdr, 2020-11-23:
Hello,

I am using microROS since two months now and I noticed some strange behaviour.

I am running the ping_pong example with freeRTOS on olimex board.
I am using UDP transport connection.

Everything runs fines in the beginning, however after some time it fails : no more ping are sent and the onboard LED does not blink anymore.

Using **ros2 topic echo /microROS/ping** I could see that it fails at arbitrary times.

I ran the test 12 times, here are the times in seconds at witch it failled : 111s, 209s, 148s, 248s, 295s, 141s, 243s, 203s, 73s, 97s, 311s, 95s

I tried also this demo in a private network with fixed IP addresses ( I changed the lwip.h file to fix the IP instead of using DHCP) but I get the same problem.

pablogs9, 2020-11-23:
Let us some time to test this.

Is this a new behavior or you have been seeing it all the time?

cocodmdr, 2020-11-23:
I have only tested this on the ping_pong demo and on my custom application which is based on ping_pong

pablogs9, 2020-11-23:
But you have had this error the two months that you have been using micro-ROS or it has suddenly appeared last week?

cocodmdr, 2020-11-23:
no since the beggining I started learning microROS

pablogs9, 2020-12-01:
Hello @cocodmdr, I'm back. I have testin micro-ROS + FreeRTOS + Olimex and I found that some defines were outdated and also that the task stack that was defined I now quite few.

I have had a problem when building with the old configuration but with the changes proposed [here](https://github.com/micro-ROS/freertos_apps/pull/43) it seems to work ok. Tomorrow I will do a long term test and I will do more improvements.

pablogs9, 2020-12-01:
Hello @cocodmdr, I have found something.

Debugging with JTAG I have found that the RTOS get stuck in [this line](https://github.com/micro-ROS/freertos_apps/blob/a5b51d8fd5c4a58f02be17cf651a4cf92483b6fd/microros_olimex_e407_extensions/Middlewares/Third_Party/FreeRTOS/Source/queue.c#L1090). The point is that this is something related to FreeRTOS + STMCubeMX and the network stack. Some other users have found the same issue: check [here](https://community.st.com/s/question/0D50X0000BJ1iquSQB/bug-in-cubemx-ide-lwip-freertos-on-nucleo-f429zi) and [here](https://forums.freertos.org/t/xsemaphoretake-function-looping-at-configassert-pxqueue-uxitemsize-0/8116).

It is possible to bypass this assert by modifying [this line](https://github.com/micro-ROS/freertos_apps/blob/a5b51d8fd5c4a58f02be17cf651a4cf92483b6fd/microros_olimex_e407_extensions/Inc/FreeRTOSConfig.h#L136) to something like:

```c
/* USER CODE BEGIN 1 */
// #define configASSERT( x ) if ((x) == 0) {taskDISABLE_INTERRUPTS(); for( ;; );} 
#define configASSERT( x )
/* USER CODE END 1 */
```

This way I have been running micro-ROS for almost an hour. But I think that disabling asserts is not the point, so I'm going to investigate more to see if there is any "proper" way of solving this.

Regarding your other issue: https://github.com/micro-ROS/freertos_apps/issues/41. I think that is the same problem so maybe we can close the other and use this one to have a discussion.

Please let us know if you find any solution.

cocodmdr, 2020-12-03:
Hello,

Thank you very much for your help !
So far it seems to work quite well, tomorrow I will try to make it run the whole day !

Regarding the othe issue I will also try tomorrow to use printf to serial on uart 6 and with transport serial on uart3.

Regards,

pablogs9, 2020-12-14:
Any update on this? Can we close?

cocodmdr, 2021-03-03:
yes, I have not had other issues with udp transport

closed 2021-03-03 by cocodmdr

ZhenshengLee, 2022-03-11:
> Thank you very much for your help !
> So far it seems to work quite well, tomorrow I will try to make it run the whole day !

Hi @cocodmdr  

Friendly ping. I've met the same problem here in #96 
Did you solve this issue?