micro-ROS/freertos_apps #41: printf to serial does not work anymore

cocodmdr, 2020-11-24:
Hello,

I am using microROS in FreeRTOS with Olimex-E407. My current installation was done the 15 October 2020.
I use serial communication with UART3 and I use UART6 as a debug console to retrieve printf messages.

To open a serial console I use this command :
- **putty /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTBT711O-if00-port0 -serial -sercfg 115200,8,n,1,N**

Everything workes fine and printf messages are being displayed.

However yesterday 23 November 2020 I installed a new microROS workspace and I tried the ping_pong demo. The demo works fine but nothing is sent to UART6 anymore.

To be sure there was nothing wrong with my installation I ran the same ping_pong demo but from my previous microROS workspace : I can see printf messages being displayed to the serial console.

So I guess there is something that went wrong between the 15th October and now.
Or is there something I am missing in order to activate the printf to serial ?

Regards,
Cocodmdr

pablogs9, 2020-11-24:
Hello @cocodmdr, I have on my TODO list to test your setup (for this and for your other issue). Give me a couple of days and I will come back with some answers.

pablogs9, 2020-12-01:
Related to https://github.com/micro-ROS/freertos_apps/issues/40

pablogs9, 2020-12-14:
Any update on this? Can we close?

cocodmdr, 2021-03-03:
yes

closed 2021-03-03 by cocodmdr