micro-ROS/freertos_apps #45: Debugging in STM32CubeIDE

Altaibaatar, 2020-12-18:
Hello

I am a newbie to ROS and currently studying it. I followed tutorials on github that you guys published. 
I want to debug it using STM32CubeIDE to the Olimex STM32F407 board. Later on, I want to try it on custom board.
I've imported Posix, Micro-xrce-dds, rcl ... libraries to my workspace. But when I try to build the project, it gives the errors related with dependancy and configurations. 
If it is possible, could you please provide a tutorial or any documentation about debugging micro-Ros app from IDE(freeRTOS). 

Kind regards

pablogs9, 2020-12-18:
Hello @Altaibaatar, can you provide some traces of your error? Currently, we are not integrated with STM32CubeIDE. Have you tried to follow the tutorial as they are? Did you achieved ROS 2 communication?

Altaibaatar, 2020-12-18:
Thank you for your reply @pablogs9  

I couldn't achieve the ROS2 communication. Recently I am failing to compile the project. Most of the errors are related to dependancies and linker errors. For example: 
`Not found file or directory: rmw_microxrcedds_c/config.h`
(Btw, I really didn't find the config.h in your repo hehe)

To build my project, I am importing and adding C source files in accordance with diagrams in the link below. As a base of my project, #https://github.com/micro-ROS/freertos_apps/tree/foxy/microros_olimex_e407_extensions is being used.
[micro-ros-on-freertos](https://www.freertos.org/2020/09/micro-ros-on-freertos.html)

About the tutorials, I followed the core tutorials that you guys are providing(Building and flashing micro-ROS to uController on Linux machines). 

Kind regards

pablogs9, 2020-12-18:
Please follow the instructions in this tutorial: https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/

And if there is any error copy-paste here the whole terminal output,

pablogs9, 2021-03-25:
Hello please check https://github.com/micro-ROS/micro_ros_stm32cubemx_utils#using-this-package-with-stm32cubeide

closed 2021-03-25 by pablogs9