micro-ROS/freertos_apps #47: Use C++ to write app

ncnynl, 2021-01-07:
I have passed the test, freertos app's ping_pong, I would like to ask if there is an example of changing app.c to app.cpp. Under keil IDE under ros1, I can integrate cpp classes into my code. But under the micro-ros, it has not been tested by adding the cpp file.

ncnynl, 2021-01-08:
At present, the problem of integrating C++ has been basically solved. Makefile adds C++ support, retains app.c and adds a func.h and func.cpp to specifically dock C++ classes. Func.cpp is used to call C++ classes and library files, and to write C standard functions, app .c Pass related parameters or perform specific operations by calling related functions of func.cpp.

In this way, the LED flicker test can now be controlled.

pablogs9, 2021-01-15:
Is this issue resolved? If so, close please

closed 2021-01-19 by ncnynl