micro-ROS/freertos_apps #48: UDP transport by ethernet port in ping_pong of olimexe407

JONDOE290, 2021-01-15:
the configure I used :
        ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp -i 192.168.0.181 -p 8888 
        ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v 6

and change the IP addresses in lwip.c:
        void MX_LWIP_Init(void)
        {
          /* Initilialize the LwIP stack with RTOS */
          tcpip_init( NULL, NULL );
          osThreadAttr_t attributes;
          ipaddr.addr = 0xC0A800B5;
          netmask.addr = 0xFFFFFF00;
          gw.addr = 0xC0A80001;
    ......

it builds and works
but win i ping 192.168.0.181,it shows:
the uart shows:
          Ethernet Initialization␍␊
           Waiting for IP␍␊
           IP: 181.0.168.192␍␊
          UDP mode => ip: 192.168.0.181 - port: 8888␊
           **********************************␊
           Task  State   Prio    Stack    Num␊
          **********************************␊
           initTask       ⇥	X⇥	23⇥	118⇥	3␍␊
           IDLE           ⇥	R⇥	0⇥	104⇥	4␍␊
          tcpip_thread   ⇥	B⇥	24⇥	149⇥	1␍␊
          microROS_app   ⇥	B⇥	25⇥	1660⇥	6␍␊
           Tmr Svc        ⇥	B⇥	2⇥	214⇥	5␍␊
          EthIf          ⇥	B⇥	48⇥	2⇥	2␍␊
          **********************************␊
          [ERROR] [ld.ld] [rclc]: [rclc_init] Error in rcl_init: error not set␊
          Failed status on line 87: 1. Aborting.␊
does olimexe407 support this configure?
or can you give me codes that works?

thank you

pablogs9, 2021-01-15:
Could you share you micro-ROS Agent output using the flag `-v6`?

JONDOE290, 2021-01-15:
> 
> 
> Could you share you micro-ROS Agent output using the flag `-v6`?

·

JONDOE290, 2021-01-15:
> 
> 
> Could you share you micro-ROS Agent output using the flag `-v6`?

[1610689478.070872] info | UDPv4AgentLinux.cpp | init | running... | port: 8888
[1610689478.071353] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 4

closed 2021-01-15 by JONDOE290