micro-ROS/freertos_apps #50: UDP transport by ethernet port in ping_pong of olimexe407 #48

JONDOE290, 2021-01-15:
the configure I used :

        ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp -i 192.168.0.181 -p 8888 
        ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v 6

and change the IP addresses in lwip.c:

        void MX_LWIP_Init(void)
        {
          /* Initilialize the LwIP stack with RTOS */
          tcpip_init( NULL, NULL );
          osThreadAttr_t attributes;
          ipaddr.addr = 0xC0A800B5;
          netmask.addr = 0xFFFFFF00;
          gw.addr = 0xC0A80001;
    ......

it builds and works
`[1610689478.070872] info | UDPv4AgentLinux.cpp | init | running... | port: 8888
[1610689478.071353] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 4`

but win i ping 192.168.0.181,it shows:
From 192.168.0.180 icmp_seq=166 Destination Host Unreachable
and win i use:ros2 topic echo /microROS/ping
WARNING: topic [/microROS/ping] does not appear to be published yet Could not determine the type for the passed topic

          Ethernet Initialization␍␊
           Waiting for IP␍␊
           IP: 181.0.168.192␍␊
          UDP mode => ip: 192.168.0.181 - port: 8888␊
           **********************************␊
           Task  State   Prio    Stack    Num␊
          **********************************␊
           initTask       ⇥	X⇥	23⇥	118⇥	3␍␊
           IDLE           ⇥	R⇥	0⇥	104⇥	4␍␊
          tcpip_thread   ⇥	B⇥	24⇥	149⇥	1␍␊
          microROS_app   ⇥	B⇥	25⇥	1660⇥	6␍␊
           Tmr Svc        ⇥	B⇥	2⇥	214⇥	5␍␊
          EthIf          ⇥	B⇥	48⇥	2⇥	2␍␊
          **********************************␊
          [ERROR] [ld.ld] [rclc]: [rclc_init] Error in rcl_init: error not set␊
          Failed status on line 87: 1. Aborting.␊

does olimexe407 support this configure?
or can you give me codes that works?

so sorry i closed the issue by accident
thank you

pablogs9, 2021-01-15:
Hello, your agent log is not set with the `-v6` flag. Please open a new agent with `-v6` flag and paste here the output after resting the board

JONDOE290, 2021-01-22:
> Hello, your agent log is not set with the `-v6` flag. Please open a new agent with `-v6` flag and paste here the output after resting the board

is this ?
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v 6
[1611297521.575553] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1611297521.575674] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4

pablogs9, 2021-01-22:
No, it can't be:

ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 **-v 6**
[1611297521.575553] info | UDPv4AgentLinux.cpp | init | running... | port: 8888
[1611297521.575674] info | Root.cpp | set_verbose_level | logger setup | **verbose_level: 4**

JONDOE290, 2021-01-23:
> No, it can't be:
> 
> ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 **-v 6**
> [1611297521.575553] info | UDPv4AgentLinux.cpp | init | running... | port: 8888
> [1611297521.575674] info | Root.cpp | set_verbose_level | logger setup | **verbose_level: 4**
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
[1611395815.655478] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1611395815.655609] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

JONDOE290, 2021-01-24:
another issue in main.c
`#ifdef MICRO_XRCEDDS_CUSTOM_SERIAL
  availableNetwork = true;
#elif defined(MICRO_XRCEDDS_UDP)
  printf("Ethernet Initialization\r\n");

	//Waiting for an IP
  printf("Waiting for IP\r\n");
  int retries = 0;
	while(gnetif.ip_addr.addr == 0 && retries < 10){
    osDelay(500);  
    retries++;
  };

  availableNetwork = (gnetif.ip_addr.addr != 0);
  if (availableNetwork){
    printf("IP: %s\r\n",ip4addr_ntoa(&gnetif.ip_addr));
  }else{
    printf("Impossible to retrieve an IP\n");
  }
#endif
`
the second part(elif...)were never used 
so i change it to
`// #ifdef MICRO_XRCEDDS_CUSTOM_SERIAL
  availableNetwork = true;
// #elif defined(MICRO_XRCEDDS_UDP)
  printf("Ethernet Initialization\r\n");

	//Waiting for an IP
  printf("Waiting for IP\r\n");
  int retries = 0;
	while(gnetif.ip_addr.addr == 0 && retries < 10){
    osDelay(500);  
    retries++;
  };

  availableNetwork = (gnetif.ip_addr.addr != 0);
  if (availableNetwork){
    printf("IP: %s\r\n",ip4addr_ntoa(&gnetif.ip_addr));
  }else{
    printf("Impossible to retrieve an IP\n");
  }
// #endif`
then is shows the above result 
why shis part can't work?

pablogs9, 2021-01-25:
I think that this is related to https://github.com/micro-ROS/freertos_apps/pull/43/files.

Please have a look to this PR, we will merge it soon. Sorry for the inconvenience.

JONDOE290, 2021-02-05:
> 
> 
> I think that this is related to https://github.com/micro-ROS/freertos_apps/pull/43/files.
> 
> Please have a look to this PR, we will merge it soon. Sorry for the inconvenience.

thanks for your help
the part works

but UDP still can't work 
and it shows same result

closed 2022-01-28 by pablogs9