micro-ROS/freertos_apps #51: Subscription callback issue [stm32 f767zi + FreeRTOS]

doodoodude, 2021-01-26:
#### Issue details
I encountered this trouble in the very first [tutorial](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/), where the ping_pong app is used as demo. But it didn't worked. Then i tried other apps (such as int32_publisher, which works well) and build my own apps. Trying to make simplified version of ping_pong app (with one subscriber and publisher and two topics), i encountered the issue mentioned in title again. 

Using HAL_TogglePin() to light LEDs for debug in my own test app, i discovered, that subscription callback hadn't called the second time. Such case: running ping_pong app on the MCU, i receive only the first ping and then the silence. 

Adding timer, which toggles the LED, in test app i discovered, that the timer 'freezes' after subscription callback executed. 

Screenshot shows, that there's respond to first message from node, but no respond to next messages (run micro_ros_agent with verbose level 5).
[image]

#### About hardware/software
- Hardware description: stm32 f767zi nucleo-144
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: Foxy
- Desktop: Ubuntu 20.04, ROS 2 Foxy

#### Flashing firmware
To be able to flash my MCU, i made some changes in 'flash.sh' script for _nucleo_f767zi_. I added the stlink-v2-1 as programmer (these lines you can also see in flash-script for _nucleo_f446ze_):
```
      elif lsusb -d 0483:374b;then
        PROGRAMMER=interface/stlink-v2-1.cfg
```

pablogs9, 2021-01-26:
Which transport are you using?

doodoodude, 2021-01-26:
> Which transport are you using?

I'm using serial port (just like in tutorial example).

pablogs9, 2021-01-26:
Let me replicate this, I'll be back in a while.

doodoodude, 2021-01-26:
Ok. I just followed the lines in the tutorial.

pablogs9, 2021-01-27:
Hello, I just have tested the `ping_pong` in the Olimex STM32-E407 and it works as expected for hours.

Maybe @jnugen, who contributed the port for f767zi can provide some information about if there is any relevant difference between these two boards.

You say that you have modified the code, is possible to share it? Which is the problem with using the very same code that we have in our repo?

doodoodude, 2021-01-27:
> You say that you have modified the code, is possible to share it?
Hi. I only modified ['flash.sh'](https://github.com/micro-ROS/micro_ros_setup/blob/foxy/config/freertos/nucleo_f767zi/flash.sh) script. Should i just attach a modified version of script to comment?

pablogs9, 2021-12-17:
Any update on this?

doodoodude, 2021-12-17:
No. I think, we can close that issue.

closed 2021-12-17 by doodoodude