micro-ROS/freertos_apps #54: Build is Broken

Kelly-Coffey, 2021-02-09:
## Issue

- Hardware description: Olimex E407
- RTOS: FreeRTOS
- Installation type: Ubuntu Tutorial Steps
- Version or commit hash: Foxy

#### Steps to reproduce the issue
Follow Tutorial at https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/

#### Expected behavior
Build completes with no errors...

#### Actual behavior
Results in this errors
/home/ubu/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:280:8: error: too many arguments to function 'uxr_init_serial_transport'
   if (!uxr_init_serial_transport(
        ^~~~~~~~~~~~~~~~~~~~~~~~~

It also fails when configured for transport UDP with similar error message 

#### Additional information
I isolated the issue when removing commits 
https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/184
https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/183

from eProsima/Micro-XRCE-DDS-Client
With these commits removed the build completes without error. 

Note: I could not remove 183 without removing 184.

pablogs9, 2021-02-10:
Hello @Kelly-Coffey, yesterday we had some problems with the build system due to some updates.

We have fixed them and I just have checked the tutorial and everything works as expected, check here:

https://asciinema.org/a/OT7pczM0tnLc21rFBtnbUbSHm

I'm closing, please do not hesitate to reopen if you have further problems.

closed 2021-02-10 by pablogs9

Kelly-Coffey, 2021-02-10:
Hi @pablogs9 , I have confirmed the serial transport works. The UDP transport is still broken.  The error is the following..

In file included from /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/include/rmw_uros/options.h:21,
                 from /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:36,
                 from /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20,
                 from /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15:
/home/ubu/microros_ws_test/firmware/mcu_ws/install/microxrcedds_client/include/uxr/client/profile/transport/custom/custom_transport.h:37:20: error: 'UCLIENT_CUSTOM_TRANSPORT_MTU' undeclared here (not in a function); did you mean 'UXR_CLIENT_CUSTOM_TRANSPORT_H_'?
     uint8_t buffer[UCLIENT_CUSTOM_TRANSPORT_MTU];
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    UXR_CLIENT_CUSTOM_TRANSPORT_H_