micro-ROS/freertos_apps #55: Build for configuration with UDP is broken

Kelly-Coffey, 2021-02-10:
Hi Pablo, et al..
## Issue template

- Hardware description: Olimex E407
- RTOS: FreeRTOS
- Installation type: Ubuntu, Foxy, 
- Version or commit hash: Foxy

#### Steps to reproduce the issue
Follow Tutorial at https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/
with the following delta

# Configure step example for UDP
ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport udp --ip 10.0.0.158 --port 8888

#### Expected behavior
Build completes with no error.

#### Actual behavior
Build fails , the first error is the following 
In file included from

 /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/include/rmw_uros/options.h:21,
                 from /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:36,
                 from /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20,
                 from /home/ubu/microros_ws_test/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15:
/home/ubu/microros_ws_test/firmware/mcu_ws/install/microxrcedds_client/include/uxr/client/profile/transport/custom/custom_transport.h:37:20: error: 'UCLIENT_CUSTOM_TRANSPORT_MTU' undeclared here (not in a function); did you mean 'UXR_CLIENT_CUSTOM_TRANSPORT_H_'?
     uint8_t buffer[UCLIENT_CUSTOM_TRANSPORT_MTU];
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    UXR_CLIENT_CUSTOM_TRANSPORT_H_

#### Additional information

Thank you

pablogs9, 2021-02-11:
Let me check

pablogs9, 2021-02-11:
Hello, sorry for that. It has been solved here: https://github.com/micro-ROS/rmw-microxrcedds/pull/102

Please close if you can run the example using UDP.

Kelly-Coffey, 2021-02-11:
I have confirmed #102 fixes UDP transport build.
Thank you
Closing

closed 2021-02-11 by Kelly-Coffey