micro-ROS/freertos_apps #58: Float64 msg Support

Kelly-Coffey, 2021-02-23:
Hello,
I have a new feature suggestion for a new example.
Do you have any examples, using any RTOS, using topics with Float64 data?

I am using the TIMx encoder hardware from the STM32F407 on the Olimex board to populate a sensor_msgs/JointState.msg. 

http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/JointState.html

The  firmware encoder data is float, but the JointState,position and JointState.velocity are float64.

Casting doesn’t to work to assign the fw encoder data to the msg data. Any suggestions how to configure to solve?

I have an imu publisher working with float data without any problems.

Thanks for your time…

pablogs9, 2021-02-23:
Hello! Based on the [int32 demo](https://github.com/micro-ROS/freertos_apps/blob/foxy/apps/int32_publisher/app.c), it should be something like:

```c
...
#include <std_msgs/msg/float64.h>
...

rcl_publisher_t publisher;
std_msgs__msg__Float64 msg;
...
void appMain(void * arg)
{
...
	RCCHECK(rclc_node_init_default(&node, "freertos_float64_publisher", "", &support));

	// create publisher
	RCCHECK(rclc_publisher_init_default(
		&publisher,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float64),
		"freertos_float64_publisher"));

...

	msg.data = (double) 3.1416;

	while(1){
		RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
		msg.data++;
	}

...
}
```

If you test it and it works fine, please do not hesitate to create a PR with this example for the rest of the community, thanks!

pablogs9, 2021-02-23:
Why does casting does not work?

JointState.velocity is an array of  doubles: float64[], have you tried to initialize it like:

```c
#define ARRAY_SIZE 1

sensor_msgs__msg__JointState msg;
msg.velocity.capacity = ARRAY_SIZE;
msg.velocity.size = ARRAY_SIZE;

// Using static memory
static double buffer[ARRAY_SIZE]
msg.velocity.data = buffer;

// Using dinamic memory
msg.velocity.data = (double*) malloc(ARRAY_SIZE*sizeof(double));

// Using just one data if ARRAY_SIZE == 1
double my_data = 3.14;
msg.velocity.data = &my_data;
```

Have you tried something like this?

Kelly-Coffey, 2021-02-23:
I have not, I will give this a try and report back. Thank you for the guidance.

Kelly-Coffey, 2021-02-23:
I verified both your static and dynamic examples compile and run. Thank you for the examples.

closed 2021-02-23 by Kelly-Coffey