micro-ROS/freertos_apps #59: How can I increase the size of the data inside a message to be published?

slahrberg, 2021-02-25:
I am using a 

Following the apps/image_publisher/ example application I want to publish audio data. For now I just use dummy data. This code will work with 1024 elements. As soon as I change this to 2014*2 or larger, the publisher will fail. RCSOFTCHECK gives me: "Failed status on line 201: 1. Continuing.". I searched for the meaning of this error code, but it seems to be a generic error, which doesn't help me. Allocating with malloc does not fail, which I am checking with the print statement.

This is how I create my message:

` 	int ELEMENTS = 1024*1;
 	msg->audio_data.capacity = ELEMENTS * sizeof(uint8_t);
 	printf("allocating %d\n", msg->audio_data.capacity);
 	msg->audio_data.data = (uint8_t *) malloc(msg->audio_data.capacity);
 	if(msg->audio_data.data == NULL){
 		printf("malloc failed!");
 	}else{
 		printf("malloc succeded!");
 	}
 	memset(msg->audio_data.data, 0x24, msg->audio_data.capacity);

 	msg->audio_data.size = msg->audio_data.capacity;
	msg->sampling_rate = 16000;
	msg->chunk_width = 1024;
	msg->sample_size = 16;`

Which is then published with a timer:

`void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
	RCLC_UNUSED(last_call_time);
	if (timer != NULL) {
		printf("Publishing some data.\n");
		RCSOFTCHECK(rcl_publish(&publisher, msg, NULL));
	}
}`

I have found other issues for example this one: https://github.com/micro-ROS/rmw-microxrcedds/issues/53 where a similar problem is being discussed. I tried changing my app-colcon.meta to set the history and MTU values with no change.

`{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=1",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=0",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=16",
                "-DRMW_UXRCE_STREAM_HISTORY=16",
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_UDP_TRANSPORT_MTU=4096",
            ]
        }
    }
}`

I have also changed a line in /firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/CMakeLists.txt so now it says : `set(UCLIENT_UDP_TRANSPORT_MTU 4096 CACHE STRING "Set the UDP transport MTU.")` . This has changed nothing.

Am I editing the wrong files or is my problem somewhere else?

pablogs9, 2021-02-26:
Which type is your message?

slahrberg, 2021-02-26:
I created a custom message type:

`uint8[] audio_data
uint16 sampling_rate
uint16 chunk_width
uint8 sample_size`

pablogs9, 2021-02-26:
Does it work if you send `audio_data` with for example 10 elements?

slahrberg, 2021-02-26:
I have tested with 3 elements and 1024, both work. Any multiplier of 1024 will not work. Btw: Is there a way to get a more precise error code from RCSOFTCHECK?

pablogs9, 2021-02-26:
Well, it should be fixed increasing `DUCLIENT_UDP_TRANSPORT_MTU` and/or `DRMW_UXRCE_STREAM_HISTORY`... the point is have you rebuilt the library?

Maybe it is still built with the default 512 B and 4 history slots, please try to rebuild the microros library from scratch and if it does not work I'll provide an example.

slahrberg, 2021-02-26:
How exactly do I rebuild the library? I have tried using `colcon build` in the main folder, which give me errors:

`$ colcon build
Starting >>> microcdr
Starting >>> microxrcedds_agent
Starting >>> drive_base_msgs
Starting >>> micro_ros_setup
Finished <<< microcdr [0.13s]                                      
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< micro_ros_setup [0.13s]
Finished <<< microxrcedds_agent [0.24s]                                   
Finished <<< rosidl_typesupport_microxrcedds_c [0.16s]
Starting >>> micro_ros_msgs
Starting >>> rosidl_typesupport_microxrcedds_cpp
Finished <<< drive_base_msgs [0.41s]                                   
Finished <<< rosidl_typesupport_microxrcedds_cpp [0.15s]
Finished <<< micro_ros_msgs [0.30s]                   
Starting >>> rmw_microxrcedds
Starting >>> micro_ros_agent
--- stderr: rmw_microxrcedds                                          
CMake Error at CMakeLists.txt:30 (find_package):
  By not providing "Findmicroxrcedds_client.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "microxrcedds_client", but CMake did not find one.

  Could not find a package configuration file provided by
  "microxrcedds_client" with any of the following names:

    microxrcedds_clientConfig.cmake
    microxrcedds_client-config.cmake

  Add the installation prefix of "microxrcedds_client" to CMAKE_PREFIX_PATH
  or set "microxrcedds_client_DIR" to a directory containing one of the above
  files.  If "microxrcedds_client" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< rmw_microxrcedds [0.25s, exited with code 1]
Aborted  <<< micro_ros_agent [0.27s]

Summary: 7 packages finished [1.01s]
  1 package failed: rmw_microxrcedds
  1 package aborted: micro_ros_agent
  2 packages had stderr output: micro_ros_agent rmw_microxrcedds
  3 packages not processed
`

pablogs9, 2021-03-04:
Any update on this?

slahrberg, 2021-03-04:
Sorry for the late answer to this issue. Yes, I got it working. I wasn't aware that i need to run the `ros2 run micro_ros_setup configure_firmware.sh` command in order for the changes to the parameters. I thought building the app itself was enough. I guess this is what you meant with "rebuilding the library".
Thank you for helping me.

closed 2021-03-04 by slahrberg