micro-ROS/freertos_apps #61: Minimum stack memory

mrbhjv, 2021-03-01:
- Hardware description: STM32F746
- Installation type: micro_ros_setup + custom_lib
- Version or commit hash: Foxy

On the application examples, I see, the stack assigned is 1500 x 4 (which is a lot). Is there a minimum suggested by you?
I've managed to make a node work with 4098 with a simple subscriber. But if I need to be able to allocate at least 4098 for each Node I want to run, it'll be kind of hard.
I guess it'll depend on the type of messages. Assume there won't be any string, no doubles, maybe floats, and no arrays of variable size. Furthermore, for the QoS, I guess I'll decrease the history to 1 and probably transient local. Anything else I should keep in mind? Is there a limit on the number of nodes I can have?

Most of the threads I have work with up to 128, while I need to increase the stack size up to at least 512 to have printf (for debug purposes).

I do know the stack size depends on the call depth, the number of arguments to methods, optimization level, etc, etc.

Thanks in advance.

pablogs9, 2021-03-01:
Some resources:
- https://micro.ros.org/docs/concepts/benchmarking/memo_prof/
- https://micro.ros.org/docs/tutorials/core/microxrcedds_rmw_configuration/

Using micro-ROS by references, decreasing the history to 1 should reduce it a bit.

mrbhjv, 2021-03-01:
I see, thanks 👍

closed 2021-03-01 by mrbhjv