micro-ROS/freertos_apps #63: error for delete_object_unlock

ncnynl, 2021-03-04:
At present, such errors often appear in the operation, leading to suspension. I want to know how to debug the problem?

show error in agent console :

`[micro_ros_agent-1] [1614827269.838182] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x5851F42D, object_id: 0x0001
[micro_ros_agent-1] [1614827269.838516] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data: 
[micro_ros_agent-1] 0000: 81 80 1D 00 05 01 06 00 01 D9 00 16 00 00
[micro_ros_agent-1] [1614827269.838615] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x5851F42D, object_id: 0x0001
[micro_ros_agent-1] 0000: 81 00 00 00 0B 01 05 00 CA 01 CA 01 80
[micro_ros_agent-1] [1614827269.855471] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x5851F42D, object_id: 0x0000
[micro_ros_agent-1] [1614827269.855631] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data: 
[micro_ros_agent-1] 0000: 81 80 1F 00 05 01 06 00 01 DB 00 01 00 00
[micro_ros_agent-1] [1614827269.855671] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
[micro_ros_agent-1] 0000: 81 00 00 00 0A 01 05 00 CB 01 00 00 80
[micro_ros_agent-1] [1614827269.855689] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
[micro_ros_agent-1] 0000: 81 00 00 00 0A 01 05 00 CB 01 00 00 80
[micro_ros_agent-1] [1614827269.855700] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
[micro_ros_agent-1] 0000: 81 00 00 00 0A 01 05 00 CB 01 00 00 80
[micro_ros_agent-1] [1614827269.857359] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
[micro_ros_agent-1] 0000: 81 00 00 00 0B 01 05 00 CA 01 CA 01 80
[micro_ros_agent-1] [1614827269.857575] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
[micro_ros_agent-1] 0000: 81 00 00 00 0A 01 05 00 CB 01 00 00 80
[micro_ros_agent-1] [1614827269.860171] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
[micro_ros_agent-1] 0000: 81 00 00 00 0A 01 05 00 20 00 00 00 80
[micro_ros_agent-1] [1614827269.862215] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 12, data: 
[micro_ros_agent-1] 0000: 81 00 00 00 03 01 04 00 00 02 FF FE
[micro_ros_agent-1] [1614827269.862371] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x5851F42D
[micro_ros_agent-1] [1614827269.862422] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x5851F42D, address: 1
`

show Error in app.c on usart:

```
[11:06:33:111] **********************************␊
[11:06:34:449] initTask       ⇥	X⇥	8⇥	1310⇥	1␍␊
[11:06:37:121] IDLE           ⇥	R⇥	0⇥	104⇥	2␍␊
[11:06:37:132] microROS_app   ⇥	B⇥	55⇥	3664⇥	4␍␊
[11:06:37:132] Tmr Svc        ⇥	B⇥	2⇥	215⇥	3␍␊
[11:06:37:132] **********************************␊
[11:06:37:369] [INFO] [ld.ld] []: Created a timer with period 500 ms.␊
[11:06:37:379] ␊
[11:06:37:445] [INFO] [ld.ld] []: Created a timer with period 500 ms.␊
[11:06:37:451] ␊
[11:06:37:518] [INFO] [ld.ld] []: Created a timer with period 500 ms.␊
[11:06:37:523] ␊
[11:07:49:870] Failed status on line 226: 1. in /home/ubuntu/microros_ws/firmware/freertos_apps/apps/ailibot2/app.c Aborting.␊

```

error  in app.c  226 line:

	//executor
	RCCHECK(rclc_executor_fini(&executor));

	//support
	RCCHECK(rclc_support_fini(&support));

	//node	
	RCCHECK(rcl_node_fini(&node));        # line 266  error here

pablogs9, 2021-03-04:
Are you deleting in the correct order?

Check [this example](https://github.com/micro-ROS/micro_ros_arduino/blob/990d60886bcc93e57d0bf4214b7431726c2fe749/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L71) for reference. If you change the order of finishing entities does it works?

```c
rcl_publisher_fini(&publisher, &node);
rcl_node_fini(&node);
rcl_timer_fini(&timer);
rclc_executor_fini(&executor);
rclc_support_fini(&support);
```

ncnynl, 2021-03-04:
I adjusted the order, but the problem was not resolved. 

The current problem may be that I added a judgment function in the while loop, which led to the judgment that the client's connection status could not be obtained correctly, so that the code behind the while was automatically executed, so the agent also automatically closed the session.

```
	while(agent_connected()){                       #  i  chage into  while(1) , it will be ok . 
		rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
	}

###
	rcl_publisher_fini(&publisher, &node);
	rcl_node_fini(&node);
	rcl_timer_fini(&timer);
	rclc_executor_fini(&executor);
	rclc_support_fini(&support);
#### after execute those line , the agent session will be deleted . 

bool agent_connected(void)
{
  bool ret ;

  if (RMW_RET_OK == rmw_uros_check_agent_status(10))
  {
    ret = true ;
  } else {
    ret = false ;
  }

  return ret ;
}
```

i can not use rmw_uros_ping_agent for check status .  i don't have this function 

@pablogs9  Thank you  for your reply

closed 2021-03-04 by ncnynl

pablogs9, 2021-03-04:
Consider using [the new ping function](https://github.com/micro-ROS/rmw-microxrcedds/blob/68adec19d27ad7d1ede92d475d59f040b4c05b6c/rmw_microxrcedds_c/include/rmw_uros/options.h#L126). In order to do so, just update your workspace or create it from scratch.