micro-ROS/freertos_apps #64: Multi threading

bjv-capra, 2021-03-08:
Hi
From the working group, I heard about adding the multithreading feature. My question is, what exactly is that you're planning to add? While working with FreeRTOS, I'd run each node on its own task, which runs as its own thread, and then use the built-in mechanism to communicate between them. What would be the case you had in mind?

pablogs9, 2021-03-08:
Hello, currently we are working on enabling the micro-ROS middleware to work in a multithreaded environment by guaranteeing a thread-safe operation between tasks in FreeRTOS and POSIX-like systems.

The first iteration will enable the possibility of running nodes, pubs, and subs in different tasks but they will still rely on a client-agent paradigm, so they will be able to communicate between them but data will be transmitted to the agent (in order to allow ROS 2 compatibility).

On a second iteration, if there is interest in this feature, we will evaluate the possibility of using OS utilities to have an interprocess communication integrated inside the micro-ROS stack that does not need the agent. But this is not in our roadmap yet. If you have special requirements you can contact us in francescafinocchiaro@eprosima.com and we can evaluate the possibility of that implementation.

pablogs9, 2021-03-08:
You can see an example of how will the API looks like here: https://github.com/micro-ROS/micro-ROS-demos/blob/feature/multithread/rclc/multithread_publisher_subscriber/main.c

This is using `pthread` but will be able to work with FreeRTOS `TaskHandle_t`. Could you provide some details about how does your use case code is organized in order to check if it will fit the design in these early development stages?

bjv-capra, 2021-03-08:
I see, in our case, we are using micro-ROS simply as a bridge between two systems. The high level and the embedded one. Although it sounds interesting to allow nodes to communicate between them natively, we're already doing it so (equivalent interprocess ) with the help of queues that have been abstracted out.
In general, we have some abstraction layers that allow us to have our system domains independent of the underlying system (OS, middleware, peripherals, etc).
I'd say it's interesting, more than anything for just starting projects that can't go through the burden of designing a good architecture to avoid interprocess problems.
Thanks for the detailed answer :+1:

closed 2021-03-08 by bjv-capra

ManvithRaj, 2022-12-06:
> Hello, currently we are working on enabling the micro-ROS middleware to work in a multithreaded environment by guaranteeing a thread-safe operation between tasks in FreeRTOS and POSIX-like systems.
> 
> The first iteration will enable the possibility of running nodes, pubs, and subs in different tasks but they will still rely on a client-agent paradigm, so they will be able to communicate between them but data will be transmitted to the agent (in order to allow ROS 2 compatibility).
> 
> On a second iteration, if there is interest in this feature, we will evaluate the possibility of using OS utilities to have an interprocess communication integrated inside the micro-ROS stack that does not need the agent. But this is not in our roadmap yet. If you have special requirements you can contact us in [francescafinocchiaro@eprosima.com](mailto:francescafinocchiaro@eprosima.com) and we can evaluate the possibility of that implementation.

hi @pablogs9 

i am interested in above second iteration you have said. is this Inter Process Communication available now ?