micro-ROS/freertos_apps #65: Create micro-ros build for STM32F407VETX

keshavchintamani, 2021-03-10:
Hi,

I have a STM32F407VETX based board and trying to find the right procedure to get a STM32CubeIDE project with micro-ROS. Not fully clear how to get the firmware and toolchain for this specific board ino the scripts. Any suggestions are really welcome!

keshavchintamani, 2021-03-10:
After doing some digging, the first step seems to add a new board to the config folder in micro_ros_setup. Thats pretty straighforward. That will pull this repo and the corresponding extensions. Whats not clear in the documentation is how the freertos extensions have been created. Is it possible to clone an extension and modify for my custom board using the STMCube IDE easily. or should I create a new STM32CubeProject for my specific board and copy over all the ROS2 specific extensions...

pablogs9, 2021-03-11:
Hello, @keshavchintamani we don't have a guide or tutorial to port new boards to FreeRTOS + micro-ROS. But I can tell you which files are added to the STCubeMX.

Using the `microros_olimex_e407_extensions` the files that are not generated from the [`micro-ROS.ioc`](https://github.com/micro-ROS/freertos_apps/blob/foxy/microros_olimex_e407_extensions/micro-ROS.ioc) are:

```
        microros_olimex_e407_extensions/FreeRTOS-Plus-POSIX/*
        microros_olimex_e407_extensions/include/*
        microros_olimex_e407_extensions/Inc/allocators.h
        microros_olimex_e407_extensions/Inc/app.h
        microros_olimex_e407_extensions/Inc/microros_transports.h
        microros_olimex_e407_extensions/Src/allocators.c
        microros_olimex_e407_extensions/Src/custom_memory_manager.c
        microros_olimex_e407_extensions/Src/microros_transports.c
        microros_olimex_e407_extensions/arm_toolchain.cmake.in
```

The important generated files that have been modified are:

- modified:   microros_olimex_e407_extensions/Inc/FreeRTOSConfig.h
<details>

```
diff --git a/microros_olimex_e407_extensions/Inc/FreeRTOSConfig.h b/microros_olimex_e407_extensions/Inc/FreeRTOSConfig.h
index 7c74a83..af7d7e8 100644
--- a/microros_olimex_e407_extensions/Inc/FreeRTOSConfig.h
+++ b/microros_olimex_e407_extensions/Inc/FreeRTOSConfig.h
@@ -44,6 +44,10 @@
 
 /* USER CODE BEGIN Includes */
 /* Section where include file can be added */
+#define configUSE_POSIX_ERRNO                   1
+#define configUSE_STATS_FORMATTING_FUNCTIONS    1
+#define configUSE_APPLICATION_TASK_TAG          1
+#define INCLUDE_xTaskGetHandle                  1
 /* USER CODE END Includes */
 
 /* Ensure definitions are only used by the compiler, and not by the assembler. */

```
</details>

- modified:   microros_olimex_e407_extensions/Makefile

<details>
```
diff --git a/microros_olimex_e407_extensions/Makefile b/microros_olimex_e407_extensions/Makefile
index ba830ff..3577571 100644
--- a/microros_olimex_e407_extensions/Makefile
+++ b/microros_olimex_e407_extensions/Makefile
@@ -20,151 +20,176 @@ TARGET = micro-ROS
 # building variables
 ######################################
 # debug build?
-DEBUG = 1
+DEBUG ?= 1
 # optimization
-OPT = -Og
+
+ifeq ($(DEBUG), 1)
+	OPT = -O0
+  	BUILD_TYPE = Debug
+else
+	OPT = -Og
+	BUILD_TYPE = Release
+endif
 
 
 #######################################
 # paths
 #######################################
[... 636 lines cut ...]
+
+  while (1){
+    if (eTaskGetState(xHandle) != eSuspended && availableNetwork){
+      HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);
+      osDelay(100);
+      HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_SET);
+      osDelay(100);
+      HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);
+      osDelay(150);
+      HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_SET);
+      osDelay(500);
+    }else{
+      HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);
+      osDelay(1000);
+      HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_SET);
+      osDelay(1000);
+    }
+  }
+  
+
   /* USER CODE END 5 */
 }
 
```
</details>

pablogs9, 2021-03-11:
Hope that this help to figure out how to modify the STCubeMX proyect to add micro-ROS

pablogs9, 2021-03-12:
Please take a look here: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils

closed 2021-03-25 by pablogs9