micro-ROS/freertos_apps #67: Can microros connect to moveit2

Needrom, 2021-04-14:
## Issue template

- Hardware description: https://github.com/Needrom/Microros-on-DIscovery-board
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
I add string subscriber to ```ping_pong``` app, and when initialize it, I get some error from the printf.

#### Expected behavior
can the microROS connect to moveit2?

#### Actual behavior
None

#### Additional information
None

pablogs9, 2021-04-14:
Yes, it can.

closed 2021-04-14 by pablogs9

Needrom, 2021-04-14:
> Yes, it can.

Does it have example for freertos_apps? I notice there is a app call ```attitude_estimator``` in ```zephyr_apps```

pablogs9, 2021-04-14:
No, there is no example in freertos_apps

Needrom, 2021-04-14:
OK

Needrom, 2021-04-15:
if I want to subscribe a ```joint_state``` topic, should I make by myself or this message is already have?

pablogs9, 2021-04-15:
You can check the micro-ROS cross-compiled packages inside `firmware/mcu_ws/install` if the message is there, you can use it.