micro-ROS/freertos_apps #71: how can I add gpio.h to app.c

Needrom, 2021-04-22:
## Issue detail

as you can see, I Can receive ```joint_states``` topic right now, but 

how can I add gpio.h from here to the app.c?

[image]

pablogs9, 2021-04-22:
Not related to micro-ROS.

closed 2021-04-22 by pablogs9

Needrom, 2021-04-22:
but for example. I add my head file in Makefile
```
#add Component file to microros system
MICROROS_INCLUDES += -I$(EXTENSIONS_DIR)/Component/A6_X4connector

```
and add header in app.c
```
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <sensor_msgs/msg/joint_state.h>
#include <stdio.h>

#include "A6_X4connector.h"

```

some thing strange thing happen
```
/home/mego/microros_ws/firmware/freertos_apps/apps/joint_states_subscriber/app.c:26:5: error: conflicting types for 'appMain'
 int appMain(void *argument)
     ^~~~~~~

```

this is the situation I don't add ```MICROROS_INCLUDES ``` line to the Makefile
```
/home/mego/microros_ws/firmware/freertos_apps/apps/joint_states_subscriber/app.c:9:10: fatal error: A6_X4connector.h: No such file or directory
 #include "A6_X4connector.h"
          ^~~~~~~~~~~~~~~~~~

```