micro-ROS/freertos_apps #72: UDP transport by ethernet doesn't work if the ethernet is not conected since the application start

Jeisongarcia9713, 2021-05-25:
- Hardware description:stm32 NUCLEO F767ZI 
- RTOS: Freertos
- Installation type:micro_ros_setup
- Version or commit hash:foxy

#### Expected behavior
the ethernet communication is available even if the ethernet is not connected when the application starts, basically the need is to have a behavior that seems to the ethernet in a ubuntu or windows device in which you could connect the ethernet in any moment and still it's going to work.

#### Actual behavior
Just if the ethernet is from the beginning connected you can unplug the ethernet and when you reconnect it is still working the communication with the agent, but if you don't plugin in the beginning you will never be able to get communication with this.

I believe this a problem related with the initialization of the ethernet, I have tried to read the code related with this but I could be able to find the piece of code that control this behavior, additionally I would like to have the DHCP client working in a task looking all the time for changes in the network, in this way in any moment of the application when I connect the ethernet the microcontroller would be able to get an IP and all the network stuff.

i would really appreciate i anyone know how to make this working and can help me. thanks.

pablogs9, 2021-05-26:
Hello @Jeisongarcia9713, sorry but NUCLEO F767ZI is a community-supported platform so we don't provide full support for all the transports layers.

Your problems seem to be related to Ethernet initialization and DHCP and I guess that maybe you will find more information looking in STM32 community forums.

In any case, if you find a solution we will be super happy to receive a PR improving the functionality of this platform.

Thanks a lot.

Jeisongarcia9713, 2021-05-27:
Thanks, i already found the way to make this work, so in these days I am going to make a PR with this.

pablogs9, 2021-05-27:
Thanks a lot!

closed 2021-05-27 by pablogs9

ZhenshengLee, 2022-03-03:
@Jeisongarcia9713  Any PR you can shared?

Thanks.