micro-ROS/freertos_apps #84: Olimex STM32 E407 FreeRTOS Broken: Build and PingPong App

Kelly-Coffey, 2021-11-07:
## Issue template

- Hardware description: Olimex E407 with Olimex Tiny Debugger
- RTOS: Foxy
- Installation type: Standard Micro-ROS Foxy Install per Tutorial
- Version or commit hash: Master/Foxy

#### Steps to reproduce the issue
Just run the FreeRTOS Ping-Pong first app Tutorial with Foxy as RTOS version

#### Expected behavior
Buld works, client can connect to agent and can send out pings

#### Actual behavior
Since last Friday 11/5 with previous install client create init options
	// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

 but agent would crash when initializing pingpong_node.
	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "pingpong_node", "", &support));  ,== crashed here.

Today following the tutorial fresh the build does not complete and issues the following error.
...
Starting >>> microxrcedds_client
Finished <<< tracetools_launch [1.02s]
Finished <<< tracetools [2.60s]
Finished <<< rosidl_parser [1.28s]                                                                    
Starting >>> rosidl_cmake
Finished <<< ros2trace [0.93s]                                                                             
Finished <<< tinydir_vendor [3.97s]                                                                         
Starting >>> rcutils
Finished <<< rosidl_cmake [1.08s]                                                                               
Starting >>> rosidl_generator_dds_idl
--- stderr: microxrcedds_client                                                                                 
/home/ubu/ws_check/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/ip_posix.c:18:10: fatal error: arpa/inet.h: No such file or directory
 #include <arpa/inet.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:391: CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/ip_posix.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:79: CMakeFiles/microxrcedds_client.dir/all] Error 2
make[1]: *** [Makefile:133: all] Error 2

#### Additional information

Thank you

pablogs9, 2021-11-10:
I have followed the steps in the [FreeRTOS tutorial](https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/) and I have found no problem. Check my [session](https://asciinema.org/a/V6tPUN8vqjpLM5R3piTIrFBg2).

Could you provide steps for replicating?

Kelly-Coffey, 2021-11-10:
Sorry. I should have been more complete with my steps to reproduce. I watched you video and was worried something was wrong with my ROS2 install until I saw you use serial as the transport. The tutorial is serial..,but I am using Ethernet as the transport. 

I found the commit in Micro-XRCE-DDS-Client 
https://github.com/eProsima/Micro-XRCE-DDS-Client/commit/b8f251d1a1010963c8a2ebdda2951fa2627820c0

"Fix missing declarations of inet_*to* family for POSIX_NOPOLL (#272)

which causes the following build error 
"--- stderr: microxrcedds_client                                          
/home/ubu/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/ip_posix.c:18:10: fatal error: arpa/inet.h: No such file or directory
 #include <arpa/inet.h>
          ^~~~~~~~~~~~~
compilation terminated."

I will continue to debug the Agent issue I am seeing with Ethernet.
Thanks

pablogs9, 2021-11-11:
I have been checking and:
- arpa/inet.h exists in the compatibility layer of LwIP for this port: check [here](https://github.com/micro-ROS/freertos_apps/blob/galactic/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/src/include/compat/posix/arpa/inet.h)
- also this folder is included in the include path: https://github.com/micro-ROS/freertos_apps/blob/84b21597234db0d2cd59a71d7139c153b2f022dd/microros_olimex_e407_extensions/Makefile#L261

Why is this not being included correctly? I'll continue checking

pablogs9, 2021-11-11:
Got it! I this that this PR can solve the problem: https://github.com/micro-ROS/freertos_apps/pull/85

Can you test this branch?

Kelly-Coffey, 2021-11-11:
I have tested. The build works and the client is able to connect and publish pings with the agent. I did not check pong.

closed 2021-11-11 by Kelly-Coffey