micro-ROS/freertos_apps #92: [BUG] Serial transport not be established if first plug the board, then start agent.

ZhenshengLee, 2022-03-03:
## Issue template

- Hardware description: stm32 NUCLEO F767ZI
- RTOS: Freertos
- Installation type: micro_ros_setup
- Version or commit hash: galactic
- transport layer: serial

#### Steps to reproduce the issue
1. compile the app int32_publisher and flash it to mcu
2. compile the agent
3. plug the board with USB
4. start agent using `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0670FF555071494867152654-if02`
5. `ros2 topic echo /freertos_int32_publisher`

#### Expected behavior

Any time you plug the board, agent would established the link.

- agent activates the serial transport, and establish a serial transport layer
- ros2 topic ehco output topic data

#### Actual behavior

agent output 

```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0670FF555071494867152654-if02 
[1646308710.731945] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1646308710.732117] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```

#### Additional information

If you first run agent, then plug the board, the link will be established, and the demo would work

I think it may be related with #72 , and may be a problem with the initialization of the serial link.

I am not familar with embedded development so any help would be appreciated.

Thanks.

pablogs9, 2022-03-03:
This is not a bug, the client will try to connect to the agent [at the begging of the app](https://github.com/micro-ROS/freertos_apps/blob/7c24435d278744f4f68de23705bf94b3b068b37e/apps/int32_publisher/app.c#L37) for a [certain amount of times](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L57) with a [certain timeout](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L58). 

If this attempt fails, in this example, the app will fail. You can check the return value of this function and retry until the agent is reached. You can [ping the agent](https://github.com/micro-ROS/micro_ros_arduino/blob/e85b9b47142601a72c742e1af71a0ff399b03c07/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L93) in order to wait until it is available. But in general, you should handle the client-agent connection.

In your use case, if you open the agent and later the client, the client might lose all the opportunities to connect. If you open the agent and then the client, it will succeed.

closed 2022-03-03 by pablogs9

ZhenshengLee, 2022-03-04:
Thanks a lot!

I will check more info from xrce-agent and xrce-client doc.