micro-ROS/freertos_apps #95: [HELP WANTED] `Failed status on line 37: 1. Aborting.` when udp connecting with stm32f767zi.

ZhenshengLee, 2022-03-10:
## Issue template

- Hardware description: nucleo-stm32f767zi connecting a router with ip 192.168.1.110
- RTOS: FreeRTOS
- Installation type: micro_ros_setup like, the src code is in https://github.com/ZhenshengLee/ros2_mcu/tree/master/libs/custom_stm32f767zi
- Version or commit hash: galactic

#### Steps to reproduce the issue
https://github.com/ZhenshengLee/ros2_mcu/tree/master/libs/custom_stm32f767zi

compile and flash with udp configuration.
colcon.meta is https://github.com/ZhenshengLee/ros2_mcu/blob/master/mcu_ws/colcon.meta
app-colcol.meta is https://github.com/ZhenshengLee/ros2_mcu/blob/master/apps/int32_publisher/app-colcon.meta

#### Expected behavior

udp connectiong success, with agent output info.

#### Actual behavior

ping 192.168.1.110 success.

but agent output nothing with

```cpp
ros2 run micro_ros_agent micro_ros_agent udp4 -p 8000 -v6
[1646914947.040102] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8000
[1646914947.040316] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```

serial output the follow

[image]

IP is got, but `Failed status on line 37: 1. Aborting.`

#### Additional information

pablogs9, 2022-03-10:
I cannot access [https://github.com/ZhenshengLee/ros2_mcu](https://github.com/ZhenshengLee/ros2_mcu/blob/master/apps/int32_publisher/app-colcon.meta). Is it private?

ZhenshengLee, 2022-03-10:
> I cannot access [https://github.com/ZhenshengLee/ros2_mcu](https://github.com/ZhenshengLee/ros2_mcu/blob/master/apps/int32_publisher/app-colcon.meta). Is it private?

Oh... @pablogs9 
You can access it now.

pablogs9, 2022-03-10:
Can you check if your code is reaching [this function](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/2195d43fff3052e3e426b8e678c682e5ffcc1cb5/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c#L76) with a JTAG debugger?

Have you tried to run a basic UDP application?

ZhenshengLee, 2022-03-11:
> Can you check if your code is reaching [this function](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/2195d43fff3052e3e426b8e678c682e5ffcc1cb5/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c#L76) with a JTAG debugger?

@pablogs9   Yes I think it's reaching the function.

[image]

I will build a minimal udp test program later.

ZhenshengLee, 2022-03-11:
@pablogs9  problem solved 

I misunderstand the meaning of `"-DRMW_UXRCE_DEFAULT_UDP_IP=192.168.1.110"`

the host(agent) should be assigned with the IP=192.168.1.110, rather than the client (stm32 device).

closing this.

Thank you.

closed 2022-03-11 by ZhenshengLee