micro-ROS/freertos_apps #96: [BUG] UDP transport with stm32f767zi hangs after a while, and network disconnected.

ZhenshengLee, 2022-03-11:
## Issue template

- Hardware description: nucleo-stm32f767zi connecting a router with ip 192.168.1.3
- RTOS: FreeRTOS
- Installation type: micro_ros_setup like, the src code is in https://github.com/ZhenshengLee/ros2_mcu/tree/master/libs/custom_stm32f767zi
- Version or commit hash: galactic

#### Steps to reproduce the issue
https://github.com/ZhenshengLee/ros2_mcu/tree/master/libs/custom_stm32f767zi

compile and flash with udp configuration.
colcon.meta is https://github.com/ZhenshengLee/ros2_mcu/blob/master/mcu_ws/colcon.meta
app-colcol.meta is https://github.com/ZhenshengLee/ros2_mcu/blob/master/apps/int32_publisher/app-colcon.meta

#### Expected behavior

ping stm32 device success, udp connectiong success, with agent output info. as long as possible.

#### Actual behavior

ping failer, and udp connection hangs, agent output hangs. After a period of time, around 500s

#### Additional information

This issue may be related with https://github.com/micro-ROS/micro_ros_setup/issues/295   https://github.com/micro-ROS/micro_ros_setup/issues/469   https://github.com/micro-ROS/freertos_apps/issues/40

the STM32 auto-generated code is by using STM32CubeMX 6.4.0 and Firmware v1.16.0

ZhenshengLee, 2022-03-11:
I tried to change `FreeRTOSConfig.h` and `main.c` accoring to https://github.com/micro-ROS/freertos_apps/issues/40#issuecomment-736597706

But did't solve that.

Acuadros95, 2022-03-11:
@ZhenshengLee Please increase the value of `INTERFACE_THREAD_STACK_SIZE` 500 here: [LWIP/Target/ethernetif.c#L42](https://github.com/ZhenshengLee/ros2_mcu/blob/master/libs/custom_stm32f767zi/LWIP/Target/ethernetif.c#L42) and give feedback.

ZhenshengLee, 2022-03-11:
> @ZhenshengLee Please increase the value of `INTERFACE_THREAD_STACK_SIZE` 500 here: [LWIP/Target/ethernetif.c#L42](https://github.com/ZhenshengLee/ros2_mcu/blob/master/libs/custom_stm32f767zi/LWIP/Target/ethernetif.c#L42) and give feedback.

@Acuadros95 
I changed `INTERFACE_THREAD_STACK_SIZE` from 350 to 1350.
It seems work. And I will give it a long-term test.

ZhenshengLee, 2022-03-12:
@Acuadros95  

24-hour int32_publisher through udp test successed. 
Closing this.

closed 2022-03-12 by ZhenshengLee