micro-ROS/freertos_apps #103: Sercvice call sometimes causes communication to stop and publisher cannot work as expected

KatterSnow, 2022-04-24:
## Issue template

- Hardware description: STM32F303 
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy
- Transport : Serial
#### Steps to reproduce the issue
Hi,
Problem 1:  

I estabulished  a micro-ROS APP with：
-1 node
-16 publishers( sum,no more than 200byte), all are best_effort
-1 subscriber( best_effort)
-2 executors: one for publishers, add a timer(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(5), publisher_timer_callback)));one for subscriber
then I set:
	while(1){
		rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(5));
		usleep(5000);
		rclc_executor_spin_some(&executor_srv, RCL_MS_TO_NS(5));
	}
         The publish frequency is no more than 60HZ (tested by "ros2 topic hz")
Then I changed my code:
	while(1){
		rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(3));
		usleep(4000);
		rclc_executor_spin_some(&executor_srv, RCL_MS_TO_NS(1));
	}
         The publish frequency is no more than 120 HZ
#### Expected behavior
The publish frequency is expected to be 200 HZ, how can I improve it? 

Problem 2:
I estabulished  a micro-ROS APP with：
-1 node
-16 publishers( sum no more than 400byte), all are best_effort
-2 services( best_effort)
-2 executors: one for publishers, add a timer(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(5), publisher_timer_callback)));one for services

ServiceTest.srv:
{
int64 a
int64 b
---
int64 sum
}

MyServiceMessage.srv
{
string demand
---
string data
}
initialized services:
	service_test__srv__ServiceTest_Request__init(&req);
    	service_test__srv__ServiceTest_Response__init(&res);
	my_service_message__srv__MyServiceMessage_Request req;
	my_service_message__srv__MyServiceMessage_Response res;
	my_service_message__srv__MyServiceMessage_Request__init(&req);
	my_service_message__srv__MyServiceMessage_Response__init(&res);
	const unsigned int SERVICE_MSG_CAPACITY = 10;
	req.demand.data = malloc(SERVICE_MSG_CAPACITY);
	req.demand.capacity = SERVICE_MSG_CAPACITY;
	snprintf(req.demand.data, req.demand.capacity, "zz1");
	req.demand.size = strlen(req.demand.data);
	
	res.data.data = malloc(SERVICE_MSG_CAPACITY);
	res.data.capacity = SERVICE_MSG_CAPACITY;
	snprintf(res.data.data, res.data.capacity, "zz2");
	res.data.size = strlen(res.data.data);

then I set:
	while(1){
		rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(3));
		usleep(4000);
		rclc_executor_spin_some(&executor_srv, RCL_MS_TO_NS(1));
	}
I try to establish the communication between client and agent, but: the service  returns 1,which means: 
Generic error to indicate operation could not complete successfully.
This problem don't always exist, I changed nothing but test again, everything can be ok  sometimes.
I modified this problem by increasing "-DRMW_UXRCE_MAX_HISTORY" :"-DRMW_UXRCE_MAX_HISTORY=3"
However, I have changed "-DUCLIENT_SERIAL_TRANSPORT_MTU=1024",
 The internal middleware layer uses a buffer of 1024x3 = 3072 Bytes, it is much larger than the data I wanted to transmit. There must be something wrong.
This buff seems to be unrelated with the attributes.stack_size in freeRTOS task， how can it be  optimized  to save space? This problem may cause RAM not enough.

Problem 3:
Test service with "ros2 service call"
For ServiceTest, everything is ok.
But for MyServiceMessage,fist time :
 I received this error:
terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException' what(): Not enough memory in the buffer stream Aborted (core dumped) 
the second time  "ros2 service call": this problem will not happen.

How can I slove this problem? it seems due to msg "string".

Problem4:
The publish frequency is no more 60 HZ which is expected to be 200 HZ

Problem5:
when I run a lot of programs in callback of service, if one node is enough to support 16 publisher and 2 services? I want to ask if micro-ros can support establish two or more tasks of freeRTOS in micro-ROS app main,or any other example to support  "multi-threading".

Problem6:
When I test service with "ros2 service call": ros2 service call -r 0.5 /service_test service_test/srv/ServiceTest "{a: 1, b: 2}"
I can get the response  several times and then there is no any response. The client-agent communication is still running.

Problem7:
Sometimes, I test service with "ros2 service call",the  client-agent communication will stop: invalid client key
The agent log:

By the way, is there any way to reduce the compile time? It cost  a lot of time to compile after I modified any code.

pablogs9, 2022-04-25:
Hello, please provide steps for replicating these problems. First of all, how are you building the micro-ROS library? AFAIK STM32F303 is not supported by micro_ros_setup. 

Also, provide a whole code that can be used to replicate this in our environment.

KatterSnow, 2022-04-25:
Hello，
Sorry for the late reply.

##How are you building the micro-ROS library?
I built the micro-ROS library with the tutorial: First micro-ROS Application on FreeRTOS 
https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/

olimex-stm32-e407 is supported by stm32f4, I change the code of it and add a new floder，like:
firmware/freertos_apps/microros_stm32_f303_extensions/
then modified the "PLATFORM": test-stm32-f303
Of course, new scripts are established in src/micro_ros_setup/config/freertos/stm-f303.
A important step: 
Library functions of stm32f303 are used to modified the "firmware/freertos_apps/microros_olimex_e407_extensions/Src/microros_transports.c"
serial transport is used as CustomTransport（baud rate： 3000000 bit/s)）.

Then, follow these steps:
ros2 run micro_ros_setup create_firmware_ws.sh freertos test-stm32-f303
ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial
ros2 run micro_ros_setup build_firmware.sh

In this way,I built the micro-ROS library(Most of files is copy from microros_olimex_e407_extensions)

Every demo of publisher/subscriber/service in "firmware/freertos_apps/apps" can work as expected.

Please provide steps for replicating these problems?

##Step1:estabulished a micro-ROS APP： firmware/freertos_apps/apps/sensor
I am sorry about that: for some reason, I cannot provide the code of all publishers.
Here,I changed my publishers.(All are imu_message now, sum no more than 400byte)

 app.c
#include <stdio.h>
#include <unistd.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rcl_action/rcl_action.h>
#include <rmw_microros/rmw_microros.h>
#include <imu_message/msg/imu_message.h>
#include "my_service_message/srv/my_service_message.h"
#include "service_test/srv/service_test.h"
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <time.h>
#ifdef ESP_PLATFORM
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#endif

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rcl_publisher_t imu_publisher1;
rcl_publisher_t imu_publisher2;
rcl_publisher_t imu_publisher3;
rcl_publisher_t imu_publisher4;
rcl_publisher_t imu_publisher5;
rcl_publisher_t imu_publisher6;
rcl_publisher_t imu_publisher7;
rcl_publisher_t imu_publisher8;
rcl_publisher_t imu_publisher9;
rcl_publisher_t imu_publisher10;
rcl_publisher_t imu_publisher11;
rcl_publisher_t imu_publisher12;
rcl_publisher_t imu_publisher13;
rcl_publisher_t imu_publisher14;
rcl_publisher_t imu_publisher15;
rcl_publisher_t imu_publisher16;
rcl_service_t service;
rcl_service_t service1;

[... 240 lines cut ...]
step7: try again: ros2 service call /mcu_service my_service_message/srv/MyServiceMessage "demand: bb"

problem3 will not happen.
How can I slove this problem? it seems due to srv "string".

step8: ros2 service call -r 0.5 /service_test service_test/srv/ServiceTest "{a: 1, b: 2}"

Then I got problem6: I can get the response several times and then there is no any response. The client-agent communication is still running.
However,sometimes the client-agent communication will stop(problem7): invalid client key

step9: I changed "service" to "subsriber",and change these codes:
     while(1){
		rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(5));
		usleep(5000);
		rclc_executor_spin_some(&executor_srv, RCL_MS_TO_NS(5));
	}
	
	The publish frequency was no more than 60HZ (tested by "ros2 topic hz")
	     while(1){
		rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(5));
		usleep(5000);
		rclc_executor_spin_some(&executor_srv, RCL_MS_TO_NS(5));
	}
	The publish frequency was no more than 120 HZ
	This is the step of problem1.

Acuadros95, 2022-04-25:
Regarding the publish frequency problem:

You configured the timer to publish each 5 ms (200 hz), but on your main loop you are adding a delay with `usleep` and  `rclc_executor_spin_some(&executor_sub, ...)`

In fact, you are getting exactly the expected rate:

- First spin (5 ms) + sleep (5 ms) + second spin (5 ms) = 15 ms ≈ 60 hz
```c
while(1) {
    rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(5));
    usleep(5000);
    rclc_executor_spin_some(&executor_sub, RCL_MS_TO_NS(5));
}
```

- First spin (3 ms) + sleep (4 ms) + second spin (1 ms) = 8 ms = 125 hz
```c
while(1) {
    rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(5));
    usleep(5000);
    rclc_executor_spin_some(&executor_sub, RCL_MS_TO_NS(5));
}
```

To solve this:
- Group both entities on the same executor and call spin once per loop, using an irrelevant sleep period on `usleep`.
- Spin each executor on a different thread.

Acuadros95, 2022-04-25:
Please check your source code carefully

- Here you are calling `rclc_service_init_best_effort` on `service` twice, leaving `service1` unitialized:
```c
RCCHECK(rclc_service_init_best_effort(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(my_service_message, srv, MyServiceMessage), "/mcu_service"));
RCCHECK(rclc_service_init_best_effort(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(service_test, srv, ServiceTest), "/service_test"));
```

- Where are `imu_publisher` and `imu_msg_pub` defined?
```c
if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&imu_publisher, &imu_msg_pub, NULL));
}
```

KatterSnow, 2022-04-25:
> Please check your source code carefully
>
> * Here you are calling `rclc_service_init_best_effort` on `service` twice, leaving `service1` unitialized:
>
> ```c [...]

KatterSnow, 2022-04-25:
> 

Thank you! "Spin each executor on a different freeRTOS task" may mean I should establish 2 tasks of freeRTOS  in  micro-ROS appMain.  I will try it!

KatterSnow, 2022-04-26:
I used one executor to add  a timer and 2 services,the publish frequency can be increased.
while(1) {
    rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(2));
    usleep(4000);
}
the publish  frequency : 85 -- 95HZ

while(1) {
    rclc_executor_spin_some(&executor_pub, RCL_MS_TO_NS(1));
    usleep(4000);
}
the publish  frequency : 61 -- 85HZ

The delay may be the intrinsic feature caused by service?

Something else:

1. problem2 still exists after increasing "-DRMW_UXRCE_MAX_HISTORY" :"-DRMW_UXRCE_MAX_HISTORY=3".
Without any change of code, 
sometimes the service returns "1" at：
RCCHECK(rclc_service_init_best_effort(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(my_service_message, srv, MyServiceMessage), "/mcu_service"));
sometimes the service returns "1" at:
RCCHECK(rclc_service_init_best_effort(&service1, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(service_test, srv, ServiceTest), "/service_test"));
sometimes everything is ok, the agent-client connection can work well.

When I set "-DRMW_UXRCE_MAX_HISTORY" :"-DRMW_UXRCE_MAX_HISTORY=2", ''-DUCLIENT_SERIAL_TRANSPORT_MTU=2048"
ERROR: region `RAM' overflowed by 1592 bytes
but When I set "-DRMW_UXRCE_MAX_HISTORY" :"-DRMW_UXRCE_MAX_HISTORY=4", ''-DUCLIENT_SERIAL_TRANSPORT_MTU=1024"
This problem doesnot exist.

2. Today, I tested micro-ROS with only 2 service: (one srv contions string)
problem3 still exists:
terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException' what(): Not enough memory in the buffer stream Aborted (core dumped)

3.  log of agent(problem7)
This problem seems only exists when involves "service" (after client-agent connecting run for a while)

1650966289.290667] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 FA 7A 07 01 08 00 7B 43 00 25 00 00 00 00
[1650966289.290865] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 FB 7A 07 01 06 00 7B 44 00 15 00 41 00 00
[1650966289.291097] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 36, data: 
0000: 81 01 FC 7A 07 01 1C 00 7B 45 00 35 00 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00
[1650966289.291267] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 20, data: 
0000: 81 01 FD 7A 07 01 09 00 7B 46 00 45 00 00 00 00 00 00 00 00
[1650966289.291473] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 20, data: 
0000: 81 01 FE 7A 07 01 0A 00 7B 47 00 55 00 00 00 00 00 00 00 00
[1650966289.291666] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 20, data: 
0000: 81 01 FF 7A 07 01 0C 00 7B 48 00 65 00 00 00 00 00 00 00 00
[1650966289.291832] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 28, data: 
0000: 81 01 00 7B 07 01 14 00 7B 49 00 75 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1650966289.292087] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 01 7B 07 01 07 00 7B 4A 00 85 00 00 00 00
[1650966289.292125] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 02 7B 07 01 07 00 7B 4B 00 95 00 00 00 00
[1650966289.292289] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 03 7B 07 01 06 00 7B 4C 00 A5 00 00 00 00
[1650966289.292580] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 04 7B 07 01 07 00 7B 4D 00 B5 00 00 00 00
[1650966289.292614] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 05 7B 07 01 06 00 7B 4E 00 C5 00 00 00 00
[1650966289.292799] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 16, data: 
0000: 81 01 06 7B 07 01 07 00 7B 4F 00 D5 00 00 00 00
[1650966289.293112] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 20, data: 
0000: 81 01 07 7B 07 01 0C 00 7B 50 00 E5 00 00 00 00 00 00 00 00
[1650966289.293165] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 40, data: 
0000: 81 01 08 7B 07 01 1D 00 7B 51 00 F5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00
[1650966289.297477] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 88, data: 
[... 107 lines cut ...]
[1650966289.327726] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 20, data: 
0000: 81 01 4F 7B 07 01 0C 00 7B 98 00 65 00 00 00 00 00 00 00 00
[1650966289.327776] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x085BAF84, len: 28, data: 
[... cut ...]
[1650966362.803445] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1650966391.735107] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1650966391.987020] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Acuadros95, 2022-05-17:
`RMW_UXRCE_MAX_HISTORY` is related to entities message queue. Check this tutorial on middleware configuration: [link](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/)

Also, make sure you are not using the same serial port for debug 
(`printf` statements, ...) and for micro-ROS communications.

pablogs9, 2022-06-14:
Any update on this?

KatterSnow, 2022-06-26:
> Any update on this?
My apologies for the delay in responding，I don't have time to go on my study on micro-ROS , because I met some big troubles. At the moment, no progress has been made.  I may work on micro-ROS  in a few months later. 
Thank you a lot for your help.!

closed 2022-06-27 by pablogs9