micro-ROS/freertos_apps #108: Why usleep(100000) is used in rclc_executor_spin_some() function?

SiddharthZokarkar, 2023-06-02:
Hardware description: ESP32
    RTOS: FreeRTOS
    Installation type: micro_ros_setup
    Version: micro-ROS running on ROS2 Foxy

Hello @pablogs9  ,

I am using the below mentioned code of int32_subscriber (below link). 
[[](url)](https://github.com/micro-ROS/freertos_apps/blob/humble/apps/int32_subscriber/app.c)

I am curious , why the delay of usleep(100000) is used inside the rclc_executor_spin_some() function?
Is there any specific reason? 

As after removing the delay the performance is better.

Thanks in advance!!

pablogs9, 2023-06-05:
Hello, @SiddharthZokarkar I do not see any usleep inside `rclc_executor_spin_some`. Could you send a link to the line?

SiddharthZokarkar, 2023-06-05:
Hello @pablogs9 ,

I think i phrased the question incorrectly . So sorry for that. 

I am talking about the usleep(100000) used inside the while loop .

while(1){
			rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
			**usleep(100000);**
	}
Line 48 in the below link:
[]((https://github.com/micro-ROS/freertos_apps/blob/humble/apps/int32_subscriber/app.c)

I am curious , why the delay of usleep(100000) is used inside the while loop?

Is there any specific reason?

This code (int32_subscriber) came as an example when I gitclone the repository
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

pablogs9, 2023-06-05:
It depends on the board and toolchain that you are using, but normally the user application shall have a priority between the inner tasks (timers, ethernet...) and the idle one.

SiddharthZokarkar, 2023-06-05:
Ok Thank you so much !!

closed 2023-06-05 by SiddharthZokarkar